Blueforcer/awtrix3 #517: Awtrix App on Android - Missing pixel in lower right corner of clock display rendering [BUG]

mleiber, 2024-03-24:
Looks like the clock display rendering in the mobile phone app (

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96
  - App version:  0.0.29 (17.03.2024)
  - Google Pixel 7 Pro running Android version 14, Build AP1A.240305.019.A1 

Steps to reproduce the behavior:

Send this to the display:

````
service: mqtt.publish
data:
  topic: awtrix_wohnhalle/notify
  payload:  "{ background: '#ff0000', text: 'abc' }"
````

The expectation would be a red square underneath the text. Instead you get a square with the right/bottom-most pixel missing.

See screenshot:

[image]

At the same time on the 'real thing', (the physical Ulanzi clock display) the red square is complete including the missing pixel.

Blueforcer, 2024-05-01:
The update is currently in the google review

Blueforcer, 2024-05-01:
It's online now. Please check it.

mleiber, 2024-05-01:
[image: image.png]

;-) all good, thx. M.

closed 2024-05-01 by Blueforcer