Blueforcer/awtrix3 #521: Bad temperature and humidity value for HTU21D and BME280 sensors.[BUG]

bladeback, 2024-03-29:
Whether I connect HTU21D or BME280, the display shows bad values ​​(low).  The actual temperature and humidity is higher.  And I even noticed that it shows the same bad values ​​for both sensors.  Can it be set somewhere?

bladeback, 2024-03-29:
Whether I connect HTU21D or BME280, the display shows bad values ​​(low).  The actual temperature and humidity is higher.  And I even noticed that it shows the same bad values ​​for both sensors.  Can it be set somewhere?

Blueforcer, 2024-03-30:
https://blueforcer.github.io/awtrix3/#/dev

closed 2024-05-01 by Blueforcer

njordan77, 2024-10-20:
I see the same @BME280. from other temp devices i see arround 24°C...but BME280 (i tried 3 different) it shows arround 15°C. I work a lot with BME280, but i never had such a bad offset. could it be that reading is done wrong?

Of course it could be offset-corrected, but it seems the values are read wrong in general. Thanks for looking into it! Norbert

Blueforcer, 2024-10-20:
The firmware uses a offset of -9 by default. Because it's made for the Ulanzi clock wich heats up itself and the sensor sits inside the case.

njordan77, 2024-10-20:
oh...sorry. that totally makes sense. they use the rubbish DHT11/22 that i never in my life managed to receive any decent values with. they are cheap but i hoped they would have taken something more mature like HTU21/si7021, not much more expensive....

Thanks for your reply. the offset makes totally sense.....its arround 24°C and leaving the system showed 14°C....so with offset of 9 its in a correct range! many thanks

njordan77, 2024-10-20:
ah sorry. i was sure it would be the DHT sensor series. SHT definitely is best/better - but as you say it will never work properly. like the smartwatches taking environementat temperature - which is impossible while wearing. Thanks for these insights and your help!