Blueforcer/awtrix3 #522: Automatic boot on "power by cable"

B42S, 2024-03-30:
# Automatic boot on "USB Power".

### Is your feature request related to a problem? New feature - not related to a problem

### Describe the solution / feature you'd like
Have TC001 boot up when USB power is applied. This is especially interesting when in has switched off due to empty akku.

### Additional context
I'm using Awtrix 3 on several TC001 as status displays for my smart home.
They are connected to USB permanently.
To save energy during the night, the house is in deep sleep mode and among others the power of the TC001 is switched off and the TC001 are sent to sleep mode (display off. When the akku is empty during the night, the TC001 power down.
In the morning the house wakes up and the USB power for the TC001 is restored. In this case I'd like to have the TC001s boot up automatically.

A switch in the dev.json would be great if possible.

Thank you in advance and a big thumbs up for your great work!
B

Blueforcer, 2024-03-30:
Duplicate of #516 and #520

closed 2024-03-30 by Blueforcer

B42S, 2024-03-30:
For that!
Thank you and have a happy Easter Weekend.

Bernhard

 

Closed #522 <https://github.com/Blueforcer/awtrix3/issues/522>  as completed.