Blueforcer/awtrix3 #524: Do not change the time format.[BUG]

bladeback, 2024-04-01:
When I change the time format to HH:MM:SS in the "on screen menu", it still shows me only HH:MM and the day of the month in a red frame on the left. How to change it to show only time with seconds?

Blueforcer, 2024-04-01:
As you can see, this format can't fit together with the calendar icon. Change the time mode to remove it. You can to this with the /settings API or with the awtrix app

bladeback, 2024-04-01:
> As you can see, this format can't fit together with the calendar icon. Change the time mode to remove it. You can to this with the /settings API or with the awtrix app

I understand that it won't fit there, I don't need the calendar icon.
How do I do this via "settings/API" please?

closed 2024-04-21 by Blueforcer

dedelastuce, 2024-04-22:
Hi I m in the same situation, I would like to remove the calendar icon to show full time (like old awtrix). Is it possible?
I tried to change time format it doesn't work calendar icon is still here

Blueforcer, 2024-04-22:
> Hi I m in the same situation, I would like to remove the calendar icon to show full time (like old awtrix). Is it possible?
> I tried to change time format it doesn't work calendar icon is still here

Both methods are described above.

on4ami, 2024-06-17:
Can someone tell me what settings requires the modification. I tried TMODE but does not solve

Blueforcer, 2024-06-17:
> Can someone tell me what settings requires the modification. I tried TMODE but does not solve

Just some line under TMODE : 
TFORMAT |  Time format for the TimeApp.

on4ami, 2024-06-18:
Hereby my settings. Calendar in front.

"MATP": true,
    "ABRI": false,
    "BRI": 125,
    "ATRANS": true,
    "TCOL": 16776954,
    "TEFF": 1,
    "TSPEED": 300,
    "ATIME": 5,
    "TMODE": 1,
    "CHCOL": 16711680,
    "CTCOL": 0,
    "CBCOL": 16777215,
    "TFORMAT": "%H:%M:%S",
    "DFORMAT": "%d/%m/%y",
    "SOM": true,
    "CEL": true,
    "BLOCKN": false,
    "MAT": 0,
    "SOUND": true,
    "GAMMA": 1.899999976,
    "UPPERCASE": true,
    "CCORRECTION": "#000000",
    "CTEMP": "#000000",
    "WD": true,
    "WDCA": 16777215,
    "WDCI": 6710886,
    "TIME_COL": 0,
    "DATE_COL": 0,
    "HUM_COL": 0,
    "TEMP_COL": 0,
    "BAT_COL": 0,
    "SSPEED": 100,
    "TIM": true,
    "DAT": false,
    "HUM": false,
    "TEMP": false,
    "BAT": false,
    "VOL": 25,
    "OVERLAY": "clear"
[image]

Blueforcer, 2024-06-18:
You cannot display this time format because it doesn't fit together with the calendar. Disable the calendar or change the time format to something without seconds.

hsauce, 2024-06-19:
Not sure why the default time format was changed without an easy to change it back to what it used to be. Now you need to know how to use the terminal or a command line prompt app (or pay for the mobile app).

Anyways here's the curl command you can use to reset it back to what it used to be. Make sure you replace `<ip-address>` with your clock's address.

```shell
curl --location 'http://<ip-address>/api/settings' \
--header 'Content-Type: application/json' \
--data '{
    "TMODE": 0
}'
```