Blueforcer/awtrix3 #525: Unable to update Firmware [BUG]

BenAhrdt, 2024-04-05:
# Bug report
Unable to update firmware.

## Describe the bug

I like to update firmware.
actual 0.94

if i click on update:

The Text: "ERR CNCT" is shown.

modem7, 2024-04-06:
Same here with 0.96

gittjoel, 2024-04-11:
did update to 0.96 in WebIf. --> OK. But if I try on the AWTRIX same error: "ERR CNCT" is shown.
Display shuts off, no reaktion anymore. 
Only switch off and on reanimates the AWTRIX.

plutosherry, 2024-04-18:
also same here with 0.96

Blueforcer, 2024-04-22:
I have now spent 3 hours trying to locate the error.
Unfortunately without success. Github refuses the connection.
Maybe someone else can try it?

modem7, 2024-04-22:
> I have now spent 3 hours trying to locate the error. Unfortunately without success. Github refuses the connection. Maybe someone else can try it?

I'm wondering if it's because you're using the raw link rather than the CDN? I suspect they probably rate limit the raw connection. 

Maybe try https://github.com/Blueforcer/awtrix3/blob/main/docs/ulanzi_flasher/firmware/firmware.bin and https://github.com/Blueforcer/awtrix3/blob/main/docs/awtrix2_flasher/firmware/firmware.bin instead in https://github.com/Blueforcer/awtrix3/blob/main/src/UpdateManager.cpp?

Blueforcer, 2024-04-22:
Both is not working.
https://github.com/Blueforcer/awtrix3/blob/main/docs/ulanzi_flasher/firmware/firmware.bin 
doesnt point directly to the blob file.

https://github.com/Blueforcer/awtrix3/raw/main/docs/ulanzi_flasher/firmware/firmware.bin
Also doesnt work

Using the API would require to expose my github token in the sourcecode.

modem7, 2024-04-22:
> Both is not working. https://github.com/Blueforcer/awtrix3/blob/main/docs/ulanzi_flasher/firmware/firmware.bin doesnt point directly to the blob file.

That was my concern, I'm wondering if adding it to releases will allow you to reference the blob itself via CDN (it seems from a very quick google search that this is the only real way of doing it from my understanding).

> Using the API would require to expose my github token in the sourcecode.

Looks like for version, you don't need to utilise the token.

```console
curl --silent "https://api.github.com/repos/Blueforcer/awtrix3/releases/latest" | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/'
0.96
```

JC-23, 2024-05-01:
The issue lies with the outdated DigiCert Root certificate used. Updating to current G2 Root solves the issue and firmware update checks work again.

BenAhrdt, 2024-05-01:
What i have to do?

Blueforcer, 2024-05-01:
> What i have to do?

Wait for the update.
However, this will only come into play in the update after next. The next update will have to be done via the app via USB or via the web interface anyway.

BenAhrdt, 2024-05-01:
OK, what is the next update? And i go to the web interface and update from there?
This works?

Blueforcer, 2024-05-01:
Next update will be 0.97?

BenAhrdt, 2024-05-16:
wich file i have to download for?

Blueforcer, 2024-05-16:
> wich file i have to download for?

For what?

BenAhrdt, 2024-05-16:
to update firmware via webpaghe?
[image]

Blueforcer, 2024-05-16:
What do you want to update?
Current firmware is 0.96. you can find all versions in the GitHub release page

BenAhrdt, 2024-05-16:
Yes, i mean... if the 0.97 is available.
Where can i find the file?
The Update from device dies not work.

modem7, 2024-05-16:
> Yes, i mean... if the 0.97 is available.
> Where can i find the file?
> The Update from device dies not work.

In the releases section of the repo.

BenAhrdt, 2024-05-16:
This file: (other version)
[image]

BenAhrdt, 2024-05-31:
when the 0.97 comes up?

Blueforcer, 2024-05-31:
> when the 0.97 comes up?

idk, But at the moment it doesn't matter, because the ondevice update will only work from 0.98 anyway. 0.97. 0.97 must be updated in a different way, via webinterface, USB or App

BenAhrdt, 2024-05-31:
ok.i updated the device to 0.96

Njdfg, 2024-08-28:
Any news when a Update is ready?

beta-j, 2024-09-19:
If you're looking for a temporary workaround to manually update Awtrix on your Ulanzi display, here's a quick summary of what you need to do:

1. First determine the IP address of your Ulanzi display 
2. Go to [https://github.com/Blueforcer/awtrix3/releases](https://github.com/Blueforcer/awtrix3/releases) and download the .bin file for the latest firmware version (eg. **ulanzi_TC001_0.96.bin**)
3. Go to http://[ULANZI_IP]/update
4. Under `Firmware`, click on `Choose File` and select the *.bin file you downloaded in Step 2.
5. Click on `Update Firmware` and wait...
6. ...Ulanzi display will automatically reboot with the updated firmware after a minute or two.

commit e44679df (2024-12-24): v0.97

- Implemented a new SSL certificate for the on-device update function. closes #525
- Resolved an issue causing the integrated icon downloader to malfunction in certain countries. closes #537 #362 #644 #642
- Added support for the Polish language by converting Polish characters to their ASCII equivalents
- MQTT Placeholders in Custom Apps
- Option to set the bar backgroundcolor in the bar drawing function.  closes #577
closed 2024-12-24 by Blueforcer