Blueforcer/awtrix3 #526: Documentation - add http method for status retrieval queries

mleiber, 2024-04-06:
In the API section of the documentation, for all but the status retrieval queries the required http method is listed. Albeit somewhat obvious, for completeness I would suggest to add a column to those 4 queries stating that the use the GET method.
Potentially also add the payload column indicating no payload required.

Blueforcer, 2024-04-21:
Doesn't "status retrieval" already say that it is a query = GET?
Of course there is no payload with mqtt because awtrix sends it on its own there is no need to send something

mleiber, 2024-04-22:
Thanks for your response. As I said above, this would only be for completeness and consistency of the documentation. Certainly not a must.