Blueforcer/awtrix3 #530: [FEATURE REQUEST] Support for polish and czech characters

ernierasta, 2024-04-14:
# Feature Request

### Is your feature request related to a problem?

When text contains polish or czech letters, they are not rendered. Biggest problem is, when I do not control source of data displayed (shared calendar for example).

### Describe the solution / feature you'd like

I would love to have possibility to display polish letters:

`ą, ć, ę, ł, ń, ó, ś, ż, ź` 

and it's upper case variants:

`Ą, Ć, Ę, Ł, Ń, Ó, Ś, Ż, Ź`

Czech letters:
`á, é, í, ó, ú, ý, č, ď, ě, ň, ř, š, ť, ž, ů`

upper case:
`Á, É, Í, Ó, Ú, Ý, Č, Ď, Ě, Ň, Ř, Š, Ť, Ž, Ů`

### Describe alternatives you've considered

If it is a hard to implement, it would help to convert for example ą -> a, č -> c. While it would be better then displaying no character at all, it would be half-solution.

BTW: Thanks all devs working on firmware, it is absolutely amazing!

Blueforcer, 2024-04-14:
There are not enough free indices in the font for more characters. Sorry.

ernierasta, 2024-04-14:
So, font definition is limited to 256 (guess) and can not be expanded? Just a wild idea, maybe we could have 2 fonts? F.e. central European languages (pl, cz, sk, ...) as separate font ... just brainstorming. What do you think?

I maybe able to create PR if there will be some consensus how to do it properly (and at the moment I do not understand how letters are generally defined).

Blueforcer, 2024-10-01:
Try the latest beta on discord