Blueforcer/awtrix3 #533: Let apps have a priority to suspend apps with lower priorities while high prio apps are active

HolgerAusB, 2024-04-17:
# Feature Request

### Is your feature request related to a problem? 

Not a problem, but it would be helpful. I just got my TC001 and it is awsome.

### Describe the solution / feature you'd like

I wrote an app with NodeRed to show which of my buddies are online at the Moment. Just a capital letter per buddy, max 6.

If at least one buddy is online, this app should stay on top and should not break by other apps with lower priority (clock, weather, etc.) but it should share the loop with apps with same priority (e.g. Doorbell rings). Doorbell app/notification shows which of my doors/doorbell rang once per ring and removed after 10 seconds.

Or explaining, the other way around:

If none of my pals is online: awtrix should change every 5 seconds between clock, weather and maybe some smart-home metrics, lets say  that is priorty=3 (standard)

When Node-red now is publishing an app with priority 1 or 2, all apps with priority 3 should stop showing-up and only the higher apps are shown on the display.

My buddy app and doorbell app would be in priority 2, while priority 1 is for the real emergencies like smoke detector or burglar alarm, water leak detection.

### Describe alternatives you've considered

I didn't use mqtt/notify yet, so I don't know if it would make sense to have priorities for notifications as well.

Maybe a prio 1 should also automatically turn on the matrix if power=off . Waking from sleep-mode is not possible, I think?

Blueforcer, 2024-05-01:
Thats not possible due the given scaffold.
You could try to modify the Apploop via API

HolgerAusB, 2024-05-01:
That would have made thing so much easier ☹️. Thanks for reply and for that great project.

closed 2024-05-01 by HolgerAusB