Blueforcer/awtrix3 #536: [FEATURE REQUEST]

ETGermany, 2024-04-21:
# Feature Request

### Is your feature request related to a problem?

The available file system is much too small

### Describe the solution / feature you'd like

I need much more memory for icons, apps, etc.

### Describe alternatives you've considered

The flash memory of TC001 is 16 MB. Why is there only 60 kiB for the file system? To use my icons, I deleted the DoNotTouch file, Ulanzi still works so far.

### Additional context

[image]

eku, 2024-04-21:
According to the [partitioning](https://github.com/Blueforcer/awtrix3/blob/main/awtrix_partition.csv)  SPIFFS is larger than 1 MB.

ETGermany, 2024-04-21:
That may be true, but my illustration shows something else. Perhaps partitioning is not working?

Blueforcer, 2024-04-21:
Dont use the original ulanzi update function. They using another weird partition table. 
If you flash it with the webflasher (with full erase) you have 1360 KB free..

ETGermany, 2024-04-21:
I had already given up with the web flasher because I was constantly getting an error message:

[image]

The EspHomeFlasher I used also deleted beforehand and output this information:

[image]

Due to your information I have now tried again, after 5 error messages it finally worked!

[image]

Great - now there is storage space.
Thank you very much!

MJHeijster, 2024-04-24:
Damn, I've been having the same issue while flashing... I have deleted a lot of icons because of the storage problem...I'll try flashing it again, but is it possible to use network storage for icons? A nas can help?

Blueforcer, 2024-04-25:
> but is it possible to use network storage for icons? A nas can help?

Thats not possible,

closed 2024-04-25 by Blueforcer