Blueforcer/awtrix3 #537: [BUG] Awtric icon install error

adrienrichmond, 2024-04-22:
# Bug report

## Describe the bug

When I try to install an impossible Metric icon: "This ID doesn't exist"

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - awtrix3 version: [ eg. v0.96 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

[image]

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2024-04-22:
https://github.com/Blueforcer/awtrix3/issues/362#issuecomment-2069952551

serfin64, 2024-05-16:
Awtrix version 0.96

[image]

Icons do not load - it says there is no such identifier

You cannot load icons and sounds into the Icons and Melodies folders through the File Manager. Uploaded only to the root directory

[image]

Blueforcer, 2024-05-16:
You can, just make sure the Path at the top is correct set to /ICONS/

TheCBiker, 2024-06-21:
The error can be caused by an adblocker or pihole. Try deactivating them temporarily.

Blueforcer, 2024-06-21:
One bug is fixed in the latest beta. You can find it in my discord channel

commit e44679df (2024-12-24): v0.97

- Implemented a new SSL certificate for the on-device update function. closes #525
- Resolved an issue causing the integrated icon downloader to malfunction in certain countries. closes #537 #362 #644 #642
- Added support for the Polish language by converting Polish characters to their ASCII equivalents
- MQTT Placeholders in Custom Apps
- Option to set the bar backgroundcolor in the bar drawing function.  closes #577
closed 2024-12-24 by Blueforcer