Blueforcer/awtrix3 #539: [BUG] After flashing the webinterface does not load the MQTT and other settings

chrisonline, 2024-04-24:
# Bug report

## Describe the bug

After flashing and connecting via AP I added the WiFi settings.
It is now connected to my WiFi, but if I open the webinterface I see no MQTT options and the "SSID"/"IP address" is empty.
See screenshot.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96 

## To Reproduce

Steps to reproduce the behavior:
Flash device and connect it via WiFi to your WiFi.
Open the webbrowser with the IP.

## Expected behavior

It should show the options MQTT and the SSID / IP address.

## Screenshots

[image]

## Logs

Attached the console log from the Chrome browser.
[192.168.1.94-1713953848159.log](https://github.com/Blueforcer/awtrix3/files/15092518/192.168.1.94-1713953848159.log)

Blueforcer, 2024-04-24:
https://github.com/Blueforcer/awtrix3/issues/511#issuecomment-1997720382

chrisonline, 2024-04-24:
Thanks that solved it. But I have never manually changed this file :-)
Anyway now it is working.

closed 2024-04-24 by chrisonline