Blueforcer/awtrix3 #540: [FEATURE REQUEST]: Listen to multiple mqtt prefixes

altepizza, 2024-04-24:
# Feature Request

### Is your feature request related to a problem?

...yes! I have multiple `Ulanzi TC001`s running at home for multiple purposes but there are certain apps and notifications that shall be pushed to all displays. Currently this leads to very unDRY implementations in my scripts and homeassistant automations :P

### Describe the solution / feature you'd like

Have the option to configure additional mqtt prefixes that `awtrix` would listen to (e.g. in my case one specific per device and a general one for all devices).
Warning: To not break existing communication from the device into mqtt, `prefix/stats` needs to run on only one prefix (the non-general topic).

### Describe alternatives you've considered

Remember all my device-prefixes and flood mqtt

Blueforcer, 2024-05-01:
Based on your reuquest, you can certainly see that this can become an endless hole once I start.  You need 2 topics in your case. the other 5.
In homeassistant it is easy to send to several devices at the same time, there is nothing "unDRY". Which also has the advantage that you can decide for each automation exactly to which it should be sent.

closed 2024-05-01 by Blueforcer

eku, 2024-05-01:
I wouldn't want to reject this so harshly. MQTT recognises a group topic. That would be enough in most cases, so not 3 or 5.

You might want to take a look at the implementation in Tasmota. The licence allows you to use it.

Blueforcer, 2024-05-01:
> I wouldn't want to reject this so harshly. MQTT recognises a group topic. That would be enough in most cases, so not 3 or 5.
> 
> You might want to take a look at the implementation in Tasmota. The licence allows you to use it.

a MQTT wildcard is not what the request creator needs.
The problem is rather that the effort for me is considerably greater, due to the current basic scaffold, than simply specifying a 2nd device for HA automation which makes it much more flexible anyway.

altepizza, 2024-05-01:
Thanks for looking into that issue. 
HA is just a minor usecase (in my case). But if it doesn't fit awtrix atm to have kind of a broadcast topic next to the device specific one, I'll just create my own little mqtt boradcaster that distributes from one topic to a list of awtrix-device-topics.