Blueforcer/awtrix3 #541: [BUG] Webinterface Timeout on two devices

Steffeng5, 2024-04-25:
# Bug report

## Describe the bug

I have two Ulanzi TC001 devices and I flashed both via Chrome Flash-Tool. Both are working more or less fine in AP mode. I can visited the web interface in AP mode and insert my WiFi Data. But after this I am not able to visit the web interface anymore. The devices do not show any error, they are working like expected, but i am not able to connect via browser to them. 

I tried downgrading to 0.94 but with the same result. Timeout. 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.94 & 0.96

## To Reproduce

Flash brand new device. Try to connect after configuring wifi

Blueforcer, 2024-04-25:
Are you still in the same network?
What IP does awtrix show while booting?

Steffeng5, 2024-04-25:
Yes Network is the correct one. It looks like if the clocks are connected via my ubiquiti AP to my network i can not access and if they are over my fritzbox in the same network everything is fine.

Is there any known issue on 2.4Ghz/5 Ghz, or sth like that?

Blueforcer, 2024-04-25:
ESP32 doesn't support 5Ghz

Steffeng5, 2024-04-25:
I updated all unifi devices and set all wifi configuration to "auto", now I can access both clocks. Great!

closed 2024-04-25 by Steffeng5

FabianTerhorst, 2024-08-03:
Having the exact same issue. I have a extra wifi network with 5ghz disabled. But the issue is happening in both wifi networks. I have multiple esp32 devices that are connecting just fine and have reachable ports. Only awtrix3 is having the issue that the port 80 is timing out. I tried erasing the awtrix but same issue. It works fine with a different router. So this seems to be a compatibility issue with awtrix3 and unifi wifi networks. A ping can't reach the awtrix3 as well.