Blueforcer/awtrix3 #544: [FEATURE REQUEST] Add 'auto' duration settings to custom app settings

mleiber, 2024-05-01:
It would be nice to have a setting to make the display duration of a text dynamic depending on content of the text. 

My first usecase would be to display song information for what's currently playing in radio or on the stereo. Song information length varies widely, some are a few characters only, some are fairly long (artist/album/title). Hard to find a duration that accomodated both, short and long.

A suggestion would be to add a mode to the duration for individual apps, example (in home assistant

```
    topic: awtrix/custom/now_playing
    payload: |-
      {
      icon: "musicvisualisation",
      pushIcon: 1,
      textCase: 2,
      duration: auto,
      repeat: 1,
      text: "{{ states('input_text.now_playing') }}"
      }
```
In `auto` mode awtrix would calculate the duration programmatically such that the text would 'fit' into one cycle of the app.

As a workaround I could calculate a duration based on speed and # of characters to display, I would think though this could be  a standard feature of awtrix3 that some users could benefit from.

Blueforcer, 2024-05-01:
Why not using the "repeat" key, I stead of duration?

Blueforcer, 2024-05-01:
Repeat will do what you want. duration overwrites the repeat key. remove the duration key and youre good to go.

mleiber, 2024-05-01:
That was quick. thanks a lot for the quick response. Deleting `duration` as we speak ;-)

closed 2024-05-01 by mleiber