Blueforcer/awtrix3 #545: Broker Mosquitto / awtrix + OpenDtu

steflp, 2024-05-01:
Hi ,  
Ive openDTU with solar panel. 
I want to send data topic from DTU to topic awtrix .
I  Dont use Home Assistant so I must find a solution homemade , i suppose.

My topics in Mqtt  : 
[image]

Thanks

PowerWiesel, 2024-05-02:
use node-red as middle man

[image]

my function node:

`var value = Math.round( msg.payload );
var unit = "Wh";

msg.payload = {  
    "text": value + '' + unit,
    "icon": "power_gen", 
    "repeat": 2, 
    "pushIcon": 0 
};

node.status({ fill: 'green', shape: 'ring', text: value + ' ' + unit});

return msg;`

steflp, 2024-05-02:
Hello,
I install  node red  : it 's OK
( I  dont know really how it s work  )  
I create  Mqtt In with my solar/ac/yieldday
I use your function (???) 
I create Mqtt out 

But after i dont know realyy how to do ....
[image]

Blueforcer, 2024-05-02:
You should join my discord for such questions. There are many users who might help you.
I think GitHub issue is not the right place for that.

steflp, 2024-05-02:
ok  ;)   sorry  ...

closed 2024-05-02 by steflp