Blueforcer/awtrix3 #547: [FEATURE REQUEST] Weekend days indicated with other color

hastaboera, 2024-05-03:
# Feature Request

### Is your feature request related to a problem?

No

### Describe the solution / feature you'd like

Add a different color to the weekend days (sat & sun) indicated on the Time app. 

### Describe alternatives you've considered

/

### Additional context

Took me a while to figure out it are the weekdays indicated as the dots. Having the weekend colored gives a quicker glance of what day of the week it is.

wolfspirituk, 2024-07-06:
Just an alternative implementation .... 
I wanted this as well - so I used an automation in Home Assistant to reset the colour with WDCA in the settings at the beginning of each day. Pick your own colours. I guess anything that can send the MQTT message or the the POST  can do the same thing.

regislutter, 2024-09-06:
I was going to request the same feature :)

regislutter, 2024-09-06:
> Just an alternative implementation .... I wanted this as well - so I used an automation in Home Assistant to reset the colour with WDCA in the settings at the beginning of each day. Pick your own colours. I guess anything that can send the MQTT message or the the POST can do the same thing.

This would only make the color of the active weekend day different. I think, like me, it's more like having the 5 days of the week in white/50%, and the weekend days in green/50% for example. Then, the active day would be white/100% or green/100% depending if it's a weekend day or not.

a-bianucci, 2024-09-16:
> Just an alternative implementation .... I wanted this as well - so I used an automation in Home Assistant to reset the colour with WDCA in the settings at the beginning of each day. Pick your own colours. I guess anything that can send the MQTT message or the the POST can do the same thing.

How did you do this?  I have been even trying to change the color of the time with TIM_COL and I can't get that to work either.

wolfspirituk, 2024-09-16:
> How did you do this? I have been even trying to change the color of the time with TIM_COL and I can't get that to work either.

I use this bit of HA YAML and call it just after midnight. There are other things in here so just look at the CHCOL WDCI and WDCA settings. This sets up a payload that I send via MQTT. You can do it in other ways - the code and data are the important bit. Works a treat.
  ma_reset_settings:
    alias: MA - Reset Settings
    fields:
      request_on:
        description: ''
    sequence:
      - variables:
          return_value:
            the_topic_root: settings
            the_payload: |-
              {
                {%- if now().weekday() >= 5 %}
                  "CHCOL": [ 0, 255, 0 ],
                  "WDCA": [ 0, 255, 0 ],
                {%- elif states('binary_sensor.workday_sensor') == 'on' %}
                  "CHCOL": [ 255, 0, 0 ],
                  "WDCA": [ 255, 0, 0 ],
                {%- else %}
                  "CHCOL": [ 0, 0, 255 ],
                  "WDCA": [ 0, 0, 255 ],
                {%- endif %}
                {%- if request_on %}
                  "BRI": 200,
                  "ABRI": false,
                {%- else %}
                  "BRI": 7,
                  "ABRI": false,
                {%- endif %}
                "WDCI": [ 64, 64, 64 ],
                "TEFF": 10,
                "BLOCKN": true,
                "TSPEED": 333,
                "UPPERCASE": false,
                "VOL": 15
              }
      - stop: Finished
        response_variable: return_value
    description: ""

HaraldGithub, 2024-10-12:
I would support this feature request! :-)

ray154856235, 2024-11-21:
I would support this feature request! :-)  
but not only weekend days also with calendar

fredbcode, 2025-05-08:
Yes great idea, I like to see the active day in a différent color