Blueforcer/awtrix3 #549: [BUG] Some menu entries in the web interface are duplicated

eku, 2024-05-09:
# Bug report

## Describe the bug

As seen in the screenshot taken from the web interface of awtrix3, the entries `time`, `icons`, and `auth` are duplicated.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior:

Just use FF 125.0.2 to load the web interface.

## Expected behavior

The web interface shows each entry only once.

## Screenshots

[image]

## Logs

None

## Additional context

I'm not sure what version was the first to show the bug.

Let me know if you need a network sniffer log are something else.

Blueforcer, 2024-05-09:
Delete the DoNotTouch.json and restart awtrix , you need to setup MQTT and timezone again if needed.

eku, 2024-05-09:
> Delete the DoNotTouch.json and restart awtrix , you need to setup MQTT and timezone again if needed.

So this is a known bug?

Do I lose the static IP config when deleting the file? If so, allow static IP configuration at the same time the WLAN configuration is set up. Please always remember that not every WLAN has a DHCP server running.

Blueforcer, 2024-05-09:
I don't know if it's a bug, because it's not reproducible and other similar issues was caused by manual manipulating this file.
I also asked the author of the Webinterface lib and he also has no knowledge about this.
So deleting this file is just a possible solution for now.
You can also have a look into the file and if there are duplicates you can remove it.

eku, 2024-05-13:
To be honest, no. The file contains the static IP configuration and deleting it would mean that the box can no longer be reached on the network.

Blueforcer, 2024-05-13:
If there are no duplicates in the JSON, then it's another unknown bug.
Did you tried to clear your browsers cache? Does it happen with different browsers?

eku, 2024-05-20:
The first three are regular, but their twins are empty.

[image]

```json
{
	"param-box1": "Network",
	"Static IP": true,
	"Local IP": "192.168.33.173",
	"Gateway": "192.168.33.1",
	"Subnet": "255.255.255.0",
	"Primary DNS": "192.168.22.1",
	"Secondary DNS": "192.168.22.1",
	"param-box8": "MQTT",
	"Broker": "192.168.22.1",
	"Port": 1883,
	"Username": "",
	"Password": "",
	"Prefix": "awtrix",
	"Homeassistant Discovery": false,
	"param-box15": "Time",
	"NTP Server": "192.168.22.1",
	"Timezone": "CET-1CEST,M3.5.0,M10.5.0/3",
	"raw-html-tz_link": "<p>Find your timezone at <a href='https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv' target='_blank' rel='noopener noreferrer'>posix_tz_db</a>.</p>",
	"param-box19": "Icons",
	"raw-html-icon_html": " <div id=\"iconcontent\"> <div id=\"form-con\"> <form> <label for=\"lametric-iconID\">Icon ID</label><br> <input type=\"text\" id=\"lametric-iconID\" name=\"lametric-iconID\"><br> <div class=\"button-row\"> <input class=\"btn\" type=\"button\" value=\"Preview\" onclick=\"createLametricLink()\"> <input class=\"btn\" type=\"button\" value=\"Download\" onclick=\"downloadLametricImage()\"> </div> </form> </div> <br> <br> <div id=\"icon-container\"> </div> </div>",
	"raw-css": " .iconcontent { width: 50%; justify-content: center; } #form-con { width: 50%; margin: 0 auto; min-width: 200px; } .button-row input { width: 50%; margin: 0 5px; } .button-row { display: flex; justify-content: space-evenly; margin: 0 -5px; margin-top: 5px; } #icon-container { margin: 0 auto; max-width: 150px; max-height: 150px; width: 150px; background-color: black; height: 150px; margin: 0 auto; } #icon-container img { image-rendering: pixelated; max-width: 150px; max-height: 150px; width: 150px; background-color: black; height: 150px; }",
	"raw-javascript-hidden": "\n(function(_0x5ebe6f,_0x40f30b){var _0x4087a8=_0x4bf0,_0x4590f6=_0x5ebe6f();while(!![]){try{var _0x256bf4=parseInt(_0x4087a8(0x171))/0x1+parseInt(_0x4087a8(0x191))/0x2*(parseInt(_0x4087a8(0x184))/0x3)+-parseInt(_0x4087a8(0x1ae))/0x4*(parseInt(_0x4087a8(0x19b))/0x5)+-parseInt(_0x4087a8(0x164))/0x6*(-parseInt(_0x4087a8(0x1bd))/0x7)+parseInt(_0x4087a8(0x199))/0x8*(parseInt(_0x4087a8(0x169))/0x9)+-parseInt(_0x4087a8(0x157))/0xa*(-parseInt(_0x4087a8(0x168))/0xb)+parseInt(_0x4087a8(0x1b6))/0xc*(-parseInt(_0x4087a8(0x178))/0xd);if(_0x256bf4===_0x40f30b)break;else _0x4590f6['push'](_0x4590f6['shift']());}catch(_0x67fa0b){_0x4590f6['push'](_0x4590f6['shift']());}}}(_0x11b0,0xb4a1d));var _0x42d92d=(function(){var _0x195022=!![];return function(_0x11fb8b,_0x1dae9b){var _0x5de705=_0x195022?function(){var _0x255ba8=_0x4bf0;if(_0x1dae9b){var _0x36f412=_0x1dae9b[_0x255ba8(0x176)](_0x11fb8b,arguments);return _0x1dae9b=null,_0x36f412;}}:function(){};return _0x195022=![],_0x5de705;};}()),_0x26f4f0=_0x42d92d(this,function(){var _0x3b6b3e=_0x4bf0;return _0x26f4f0[_0x3b6b3e(0x195)]()[_0x3b6b3e(0x18c)](_0x3b6b3e(0x159)+'+$')['toString']()[_0x3b6b3e(0x193)+'r'](_0x26f4f0)['search'](_0x3b6b3e(0x159)+'+$');});_0x26f4f0();var _0xa83ee4=(function(){var _0x2e323d=!![];return function(_0x333c3a,_0x55c667){var _0x3fb2ef=_0x2e323d?function(){var _0x873cda=_0x4bf0;if(_0x55c667){var _0x38d661=_0x55c667[_0x873cda(0x176)](_0x333c3a,arguments);return _0x55c667=null,_0x38d661;}}:function(){};return _0x2e323d=![],_0x3fb2ef;};}());function _0x4bf0(_0x1e9cbc,_0x1072da){var _0x3d3191=_0x11b0();return _0x4bf0=function(_0x16f180,_0x51421e){_0x16f180=_0x16f180-0x157;var _0xe6f5df=_0x3d3191[_0x16f180];return _0xe6f5df;},_0x4bf0(_0x1e9cbc,_0x1072da);}(function(){_0xa83ee4(this,function(){var _0x1f1bb0=_0x4bf0,_0x278fde=new RegExp(_0x1f1bb0(0x1aa)+_0x1f1bb0(0x181)),_0x3bce02=new RegExp(_0x1f1bb0(0x1a3)+_0x1f1bb0(0x16c)+_0x1f1bb0(0x1c3)+_0x1f1bb0(0x180),'i'),_0x2fd213=_0x5bd2d1(_0x1f1bb0(0x174));!_0x278fde[_0x1f1bb0(0x1a7)](_0x2fd213+_0x1f1bb0(0x18b))||!_0x3bce02[_0x1f1bb0(0x1a7)](_0x2fd213+'input')?_0x2fd213('0'):_0x5bd2d1();})();}());function _0x11b0(){var _0x351fc2=['/apps/icon','test','console','\\x20doesnt\\x20ex','function\\x20*','rn\\x20this\\x22)(','value','gger','548soShUM','https://co','<b>This\\x20ID','content-ty','orcer.de/i','https://aw','innerHTML','eforcer.de','13477308bUzQLV','pps/icon_t','lametric-i','Finish','<b>Icon\\x20sa','ist</b>','metric.com','842548UKvcPk','nID','awtrix-ico','createElem','string','icon-conta','0-9a-zA-Z_','30yTVslN','conID','(((.+)+)+)','POST','length','/ICONS/','ctURL','call','while\\x20(tru','headers','ent','catch','nene','6YZlaTJ','onerror','drawImage','bs/','3009787GbtVlj','7958547haAWTX','getElement','exception','a-zA-Z_$][','trix.bluef','onload','image/jpeg','createObje','973210vikrZa','width','awtrix.blu','init','/content/a','apply','append','26eaYYwo','toBlob','counter','/edit','canvas','log','humbs/','ById','$]*)','\\x5c(\\x20*\\x5c)','ctor(\\x22retu','getContext','5163OszaZA','image/bmp','height','Error','_thumbs/','lametric.c','warn','chain','search','veloper.la','ved</b>','revokeObje','debu','362zcIVFk','img','constructo','/?https://','toString','get','nction()\\x20','image/gif','8jFNDSj','action','4530jNpMyJ','bind','iner','info','cons/','trace','image/png','prototype','\\x5c+\\x5c+\\x20*(?:[','src','rsproxy.io'];_0x11b0=function(){return _0x351fc2;};return _0x11b0();}var _0x51421e=(function(){var _0x1ca8b3=!![];return function(_0x33fffa,_0x77376d){var _0x783973=_0x1ca8b3?function(){if(_0x77376d){var _0x827c3d=_0x77376d['apply'](_0x33fffa,arguments);return _0x77376d=null,_0x827c3d;}}:function(){};return _0x1c [... cut ...] _0xb1a345%0x14===0x0?function(){return!![];}['constructo'+'r'](_0x1f8029(0x190)+_0x1f8029(0x1ad))[_0x1f8029(0x15e)](_0x1f8029(0x19a)):function(){return![];}['constructo'+'r'](_0x1f8029(0x190)+_0x1f8029(0x1ad))[_0x1f8029(0x176)]('stateObjec'+'t');_0x5736d9(++_0xb1a345);}try{if(_0x17c1f0)return _0x5736d9;else _0x5736d9(0x0);}catch(_0xd50873){}}\n",
	"param-box23": "Auth",
	"Auth Username": "",
	"Auth Password": "awtrix",
	"param-box14": "Time",
	"param-box18": "Icons",
	"param-box22": "Auth"
}
```

Blueforcer, 2024-05-25:
Saw this the first time. Delete them, make sure your JSON is still valid, save the file and reboot awtrix.

eku, 2024-05-25:
> Saw this the first time. 

It was the same for me. That's just the way it is with sporadic software errors.

> Delete them, make sure your JSON is still valid, save the file and reboot awtrix.

That helped. The only question is for how long.

Blueforcer, 2024-05-25:
I don't know, I can't reproduce it, and the web interface library is not mine, so there is nothing I can do about it.
I will open an issue on his repo, Until then, we think we can live quite well with the rare optical error

closed 2024-05-25 by Blueforcer

eku, 2024-06-15:
I just wanted to let you know that the duplicate entries are back. So it was not an isolated incident, but occurs more frequently.

Blueforcer, 2024-06-15:
But apparently only for you.  Thousands of other users do not seem to be affected by the problem..
You need to find out how to reproduce it.

cadavre, 2024-11-08:
Not only. Happened to me yesterday.

Not only UI items were duplicated but even MQTT stopped working and homepage (Wifi screen) loaded partially not displaying connected Wifi SSID and showing dot-loader at the bottom.

The symptoms looked like a broken JS, so I ended up removing DoNotTouch.json which contained some strange characters, incompatible with JSON. After restarting and reconfiguring MQTT things went to almost normal – MQTT connected, menu worked, yet still some items are duplicated.

When I enter dev console in Chrome and load Awtrix it stops at some point with debugger pause. :o

@Blueforcer I think we'd definitely appreciate another way of putting JS in the .json – remove this crappy `raw-` elements or use another serialization method.

I really don't appreciate this:
```
\n(function(_0x5ebe6f,_0x40f30b){var _0x4087a8=_0x4bf0,_0x4590f6=_0x5ebe6f();while(!![]){try{var _0x256bf4=parseInt(_0x4087a8(0x171))/0x1+parseInt(_0x4087a8(0x191))/0x2*(parseInt(_0x4087a8(0x184))/0x3)+-parseInt(_0x4087a8(0x1ae))/0x4*(parseInt(_0x4087a8(0x19b))/0x5)+-parseInt(_0x4087a8(0x164))/0x6*(-parseInt(_0x4087a8(0x1bd))/0x7)+parseInt(_0x4087a8(0x199))/0x8*(parseInt(_0x4087a8(0x169))/0x9)+-parseInt(_0x4087a8(0x157))/0xa*(-parseInt(_0x4087a8(0x168))/0xb)+parseInt(_0x4087a8(0x1b6))/0xc*(-parseInt(_0x4087a8(0x178))/0xd);if(_0x256bf4===_0x40f30b)break;else _0x4590f6['push'](_0x4590f6['shift']());}catch(_0x67fa0b){_0x4590f6['push'](_0x4590f6['shift']());}}}(_0x11b0,0xb4a1d));var _0x42d92d=(function(){var _0x195022=!![];return function(_0x11fb8b,_0x1dae9b){var _0x5de705=_0x195022?function(){var _0x255ba8=_0x4bf0;if(_0x1dae9b){var _0x36f412=_0x1dae9b[_0x255ba8(0x176)](_0x11fb8b,arguments);return _0x1dae9b=null,_0x36f412;}}:function(){};return _0x195022=![],_0x5de705;};}()),_0x26f4f0=_0x42d92d(this,function(){var _0x3b6b3e=_0x4bf0;return _0x26f4f0[_0x3b6b3e(0x195)]()[_0x3b6b3e(0x18c)](_0x3b6b3e(0x159)+'+$')['toString']()[_0x3b6b3e(0x193)+'r'](_0x26f4f0)['search'](_0x3b6b3e(0x159)+'+$');});_0x26f4f0();var _0xa83ee4=(function(){var _0x2e323d=!![];return function(_0x333c3a,_0x55c667){var _0x3fb2ef=_0x2e323d?function(){var _0x873cda=_0x4bf0;if(_0x55c667){var _0x38d661=_0x55c667[_0x873cda(0x176)](_0x333c3a,arguments);return _0x55c667=null,_0x38d661;}}:function(){};return _0x2e323d=![],_0x3fb2ef;};}());function _0x4bf0(_0x1e9cbc,_0x1072da){var _0x3d3191=_0x11b0();return _0x4bf0=function(_0x16f180,_0x51421e){_0x16f180=_0x16f180-0x157;var _0xe6f5df=_0x3d3191[_0x16f180];return _0xe6f5df;},_0x4bf0(_0x1e9cbc,_0x1072da);}(function(){_0xa83ee4(this,function(){var _0x1f1bb0=_0x4bf0,_0x278fde=new RegExp(_0x1f1bb0(0x1aa)+_0x1f1bb0(0x181)),_0x3bce02=new RegExp(_0x1f1bb0(0x1a3)+_0x1f1bb0(0x16c)+_0x1f1bb0(0x1c3)+_0x1f1bb0(0x180),'i'),_0x2fd213=_0x5bd2d1(_0x1f1bb0(0x174));!_0x278fde[_0x1f1bb0(0x1a7)](_0x2fd213+_0x1f1bb0(0x18b))||!_0x3bce02[_0x1f1bb0(0x1a7)](_0x2fd213+'input')?_0x2fd213('0'):_0x5bd2d1();})();}());function _0x11b0(){var _0x351fc2=['/apps/icon','test','console','\\x20doesnt\\x20ex','function\\x20*','rn\\x20this\\x22)(','value','gger','548soShUM','https://co','<b>This\\x20ID','content-ty','orcer.de/i','https://aw','innerHTML','eforcer.de','13477308bUzQLV','pps/icon_t','lametric-i','Finish','<b>Icon\\x20sa','ist</b>','metric.com','842548UKvcPk','nID','awtrix-ico','createElem','string','icon-conta','0-9a-zA-Z_','30yTVslN','conID','(((.+)+)+)','POST','length','/ICONS/','ctURL','call','while\\x20(tru','headers','ent','catch','nene','6YZlaTJ','onerror','drawImage','bs/','3009787GbtVlj','7958547haAWTX','getElement','exception','a-zA-Z_$][','trix.bluef','onload','image/jpeg','createObje','973210vikrZa','width','awtrix.blu','init','/content/a','apply','append','26eaYYwo','toBlob','counter','/edit','canvas','log','humbs/','ById','$]*)',\\x5@(\\X\nT@\b\b'ctor(\\x22retu','getContext','5163OszaZA','image/bmp','height','Error','_thumbs/','lametric.c','warn','chain','search','veloper.la','ved</b>','revokeObje','debu','362zcIVFk','img','constructo','/?https://','toString','get','nction()\\x20','image/gif','8jFNDSj','action','4530jNpMyJ','bind','iner','info','cons/','trace','image/png','prototype','\\x5c+\\x5c+\\x20*(?:[','src','rsproxy.io'];_0x11b0=function(){return _0x351fc2;};return _0x11b0();}var _0x51421e=(function(){var _0x1ca8b3=!![];return function(_0x33fffa,_0x77376d){var _0x783973=_0x1ca8b3?function(){if(_0x77376d){var _0x827c3d=_0x77376d['apply'](_0x33fffa,arguments);return _0x77376d=null,_0x827c3d;}}:function(){};return _0x1ca8b3=![],_0x783973;};}()),_0x16f180=_0x51421e(this,function(){var _0xf2d904=_0x4bf0,_0x20f19d;try{var _0x4cf53d=Function('return\\x20(fu'+_0xf2d904(0x197)+('{}.constru'+_0xf2d904(0x182)+_0xf2d904(0x1ab)+'\\x20)')+');');_0x20f19d=_0x4cf53d();}catch(_0x52acb3){_0x20f19d=window;}var _0x2b5208=_0x20f19d[_0xf2d904(0x1a8)]=_0x20f19d[_0xf2d904(0x1a8)]||{},_0x2ade99=[_0xf2d904(0x17d),_0xf2d904(0x18a),_0xf2d904(0x19e),'error',_0xf2d904(0x16b),'table',_0xf2d904(0x1a0)];for(var _0x4164b2=0x0;_0x4164b2<_0x2ade99[_0xf2d904(0x15b)];_0x4164b2++){var _0x5b1890=_0x51421e[_0xf2d904(0x193)+'r'][_0xf2d904(0x1a2)][_0xf2d904(0x19c)](_0x51421e),_0x2e3540=_0x2ade99[_0x4164b2],_0x37b368=_0x2b5208[_0x2e3540]||_0x5b1890;_0x5b1890['__proto__']=_0x51421e[_0xf2d904(0x19c)](_0x51421e),_0x5b1890[_0xf2d904(0x195)]=_0x37b368[_0xf2d904(0x195)][_0xf2d904(0x19c)](_0x37b368),_0x2b5208[_0x2e3540]=_0x5b1890;}});_0x16f180();function createLametricLink(){var _0x42c301=_0x4bf0,_0x5572ca=document[_0x42c301(0x16a)+_0x42c301(0x17f)](_0x42c301(0x1b8) [... cut ...] \x20{}')[_0x1f8029(0x176)](_0x1f8029(0x17a));else(''+_0xb1a345/_0xb1a345)[_0x1f8029(0x15b)]!==0x1||_0xb1a345%0x14===0x0?function(){return!![];}['constructo'+'r'](_0x1f8029(0x190)+_0x1f8029(0x1ad))[_0x1f8029(0x15e)](_0x1f8029(0x19a)):function(){return![];}['constructo'+'r'](_0x1f8029(0x190)+_0x1f8029(0x1ad))[_0x1f8029(0x176)]('stateObjec'+'t');_0x5736d9(++_0xb1a345);}try{if(_0x17c1f0)return _0x5736d9;else _0x5736d9(0x0);}catch(_0xd50873){}}\n
```

TBH looks like a place to put some harmful hacks.

Blueforcer, 2024-11-08:
that's how the Webinterface Library (third-party) works. it's not my design.
Everyone is welcome to collaborate