Blueforcer/awtrix3 #550: [BUG] Security problem : nmap cause crash

Jezza34000, 2024-05-10:
# Bug report

## Describe the bug

Using this command from awtrix network :
`nmap -p- <AWRIX3_IP>`

Cause awtrix3 boot loop

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Try `nmap -p- <AWRIX3_IP>`

## Expected behavior

Scanner start, after 10 seconds Awtrix will reboot.

Blueforcer, 2024-05-10:
What is the security problem here?
The firmware is open source, you can see every Service in the code.

Running a Portscanner on Awtrix is quite a rare usecase. And I'm pretty sure the crash is not related to the Firmware code but to core functionalities like the wifi, mdns or mqtt library

Jezza34000, 2024-05-10:
Yes it's not security problem
Ok for information

closed 2024-06-11 by Blueforcer