Blueforcer/awtrix3 #554: [BUG] Documentation for Power and Sleep works only with: [IP]/power and [IP]/sleep iso [IP]/api/power and [IP]/api/sleep

fmdvries, 2024-05-15:
# Bug report

## Describe the bug

With my Ulanzi together with HomeAssistant and Node-Red, I can only get Power/Sleep working correctly with the following endpooints:

[IP]/power and [IP]/sleep 

The documentation says:

[IP]/api/power and [IP]/api/sleep

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

## To Reproduce

Steps to reproduce the behavior:

Send MQTT message via a create message node:

msg.payload = {"power": true};
return msg;

and publish via an MQTT publish node with topic:

[IP]/api/power as in documentation --> no effect

and:

[IP]/power as in documentation --> works

Same for sleep.

## Expected behavior

Power and Sleep to work according to the documentation (api.md)

So proposed solution: validate and update documentation to exclude the /api part in the Topic/URL

## Screenshots

N/A

## Logs

N/A

## Additional context

N/A

Blueforcer, 2024-05-16:
Some things in your report are strange.
**[IP]/power** does not exist for mqtt because MQTT does not use IP
Do not confuse the MQTT topic with the HTTP base URL
[image]

fmdvries, 2024-05-16:
Ah you're completely right! My bad, I just didn't look carefully :-(

closed 2024-05-16 by fmdvries