Blueforcer/awtrix3 #556: [BUG] half of bottom row pushed one pixel to the left

uuzi, 2024-05-22:
Half of the bottom most row of pixels ist pushed one pixel to the left after flashing. Tried flashing again, same result.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96
[image]
[image]

Blueforcer, 2024-05-22:
You're the first one with this. So I assume this is a hardware fault. 
Flash the original firmware and see if it is persistent

uuzi, 2024-05-23:
Will do and report back this evening.

closed 2024-05-23 by uuzi

L1ttl3b4k3r, 2025-08-05:
> Will do and report back this evening.
> 
> I just checked and yes it seems like a hardware fault. I've contacted Ulanzi support, let's see what they say.

I recently encountered exactly the same issue. What was the reply of Ulanzi?
What was the cure?

L1ttl3b4k3r, 2025-08-06:
Did you sent them pictures with the original firmware or with awtrix3 ?

uuzi, 2025-08-07:
Just to be safe, I first returned to original firmware.