Blueforcer/awtrix3 #558: [BUG] API doesn't work

Goldsucher, 2024-05-26:
# Bug report

## Describe the bug

I have a fresh installation of awtrix on my ulanzi clock but the API doesn't work. If I try

` curl -d '{"force":true,"text":"Awesome","icon":1,"color":[0,255,255],"count":2}' -H "Content-Type: application/json" -X POST http://192.168.250.60:7000/api/v3/notify `

I get `curl: (7) Failed to connect to 192.168.250.60 port 7000 after 135 ms: Couldn't connect to server`

I really have no idea why this is the case. I can connect with the Awtrix iOS App and mqtt is also working.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Install Awtrix. Send curl / http request

## Expected behavior

Text is shown on clock

Blueforcer, 2024-05-26:
Port 7000 doesn't exist.
Where did you get this information?

Blueforcer, 2024-05-26:
Some json keys doesn't exist and also port 7000 doesn't exist. Please stick to the documentation

Goldsucher, 2024-05-26:
sorry but that is exactly what your documentation says:
https://awtrixdocs.blueforcer.de/#/en-en/api

Blueforcer, 2024-05-26:
That's a different and old project. It's not linked to this repo in any way. Use the documentation of this AWTRIX 3 Repo

Goldsucher, 2024-05-26:
Oh sorry, I must have mixed it up. Thanks for the clarification! 🙏

closed 2024-05-26 by Goldsucher