Blueforcer/awtrix3 #560: [BUG] - Icon not animating correctly

wolfspirituk, 2024-06-01:
# Bug report

This is the icon I created and want to use:
[image]
This is from Lametric site. It is animated.

The last frame is corrupted on the Awtrix.
It happens with most of the animated gifs I've tried. Most were created by others on Lametric. e.g. 9521

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version:  v0.96

## To Reproduce

0) Search for icon 60748 on Lametric website - hover over it - to see what should happen.

1) Preview icon 60748 on Awtrix webUI. The preview seems OK.

2) Download it via the WebUI

3) Goto the file manager in Awrtrix web UI. click the icon folder and then 60748.gif 
It is different from the one seen on the Lametric website (where the icon was created)
[image]
There is no frame with a door and window in white

4) Use the icon in an "app"
The first frames work OK - but the last one is corrupted.
OK: [image]
Corrupted: [image]
There is never a brown roof or gray building in the original.

I've tried reloading the same gif - no difference.

## Expected behavior

The Lametric site has the correct animation - if you hover over the icon it animates.

## Screenshots

## Logs

## Additional context

RAM isn't an issue - never goes below 105k (40k seems to be reported elsewhere as a problem)
[image]

I think I am only using MQTTs to operate the "clock".

It crashes or shuts down irregularly every few hours.
For some reason it use to be regular at almost exactly 4 hours (I don't think I was causing that in code, but might have)
[image]

I can reasonably reliably crash the clock if I hit is with lots of MQTTs for new apps over a few seconds. Maybe 30-40 requests with a delete payload ("") and then the new data.

wolfspirituk, 2024-06-01:
Forgot to say - the 30-40 requests are triggered via a HASS dashboard automation - it turns the apps on or off - I have 20 odd apps and this crash seems to occur soon after I turn them all off one by one and then on again one by one. As quick as my finger can click the dashboard buttons.

Blueforcer, 2024-06-01:
Not every gif format is compatible, see
https://blueforcer.github.io/awtrix3/#/icons?id=important-note

wolfspirituk, 2024-06-01:
Thanks! I'll try to find a tool that doesn't bloat and fuzz the image.

closed 2024-06-01 by wolfspirituk

Blueforcer, 2024-06-01:
Did you find the error? Would be helpful for other users if my comment doesn't help. Greetings

Blueforcer, 2024-06-01:
You should open another issue for the additional context, since the crash has nothing to do with the icon colors.

wolfspirituk, 2024-06-01:
No, not yet. The tool mentioned bloats the size due to poor "compression" I guess! It then fails to load in the Awtrix interface with ERROR 500.
I'll try to find another - but might just be quicker to create again.
However I think you are right as the windows and door could have transparancy in it as the Lametric interface did it when you click a pixel the 2nd time rather than painting with a black pixel - no indication in the UI!
Judging by the number of icons that do similar - other people have created them that way as well.

I thought it might have been corruption linked to the crashes.
Thanks for input (and Awtrix) - I'll raise the crash issue separately - but I can't think how to give you enough information to resolve it.

wolfspirituk, 2024-06-01:
Actually - if an icon has transparency - will it be handled safely ?
Of course I might not know they have transparency - especially as the LaMetric tool can add transparency to the icon.
Also is there an easy way of mass-deleting ? - the UI goes back to root on each delete.

wolfspirituk, 2024-06-15:
This is a comment to help others facing this issue.... as best as I can tell what is happening....

Transparency doesn't seem to be the issue as some icons that seem to have transparency work OK. e.g. 22003 (the icon has a grey background in the file manager view on Awtrix but are produced with off/black pixels on the awtrix display)

The colour table seems to be the issue and fixing it with this tool: https://ezgif.com/repair is a solution.
Repair the gif with "Reset color tables" then they work fine with the Awtrix. The last 3 of the 4 options all seemed to work - I just like the sound of the last one!

I have no idea what is happening behind the scenes and why the LaMetric icon tool creates something that Awtrix doesn't handle.

Other transparency stripping tools I tried seem to do some other compression / fuzzing algo - that might be great in a 200x200 gif but not desirable in a 8x8 one!