Blueforcer/awtrix3 #562: [BUG] Matrix orientation for custom 32x8 matrix panel with ESP32

AWSW-de, 2024-06-03:
Hello =)

First of all: This is more a question than a bug and i want to say, that i am a BIG fan of this projekt! :)

Because i like to build my own hardware, i made a first variant of a new 32x8 matrix NeoPixel 3D printable model, that does not even need soldering parts. I then added a super simple digital clock code to it like shown here: 
https://www.printables.com/de/model/896884-pixelclock-32x8

Before i found this project here i wanted to build something similar to show some information about my smart home, calendar events and so on, on such a display which i plan to create much larger too, but hat is a different topic for a different time... 

As shown in the pictures below and in the link the general function of the matrix works fine, but i think there is a problem with the matrix orientation of this special NeoPixel matrix. I think the "zigzack" of the LEDs is different to the ones normally used here.

For the simple digital clock is used this to initialize the matrix:

`Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(32, 8, LEDPIN, NEO_MATRIX_TOP + NEO_MATRIX_LEFT + NEO_MATRIX_COLUMNS + NEO_MATRIX_ZIGZAG, NEO_GRB + NEO_KHZ800);`

Maybe this helps to compare the different settings.

Can you please give me hint to this? I had a quick look at a few of your source files i expected the setting in but was not able to find it.

Thanks in advance and maybe this opens some oppertunities for your great project to run on even more devices soon. =) 

## Additional information

- Devices involved:
  - Model: Custom: https://www.printables.com/de/model/896884-pixelclock-32x8
  - Matrix: https://www.amazon.de/dp/B09Z6SJFLX
  - ESP32: https://www.amazon.de/dp/B0CH2SHL8S
  - awtrix3 version: 0.96

## Screenshots

[image]

[image]

AWSW-de, 2024-06-03:
Hello,
just found the solution by creating a dev.json file with this content:

[image]

Thanks a lot =)

Blueforcer, 2024-06-03:
Super!
Bist herzlich eingeladen meinem discord zu joinen :)

closed 2024-06-03 by Blueforcer

AWSW-de, 2024-06-03:
Danke Dir! Das mach ich gerne. =) 

PS: Vielleicht könntet ihr ein paar meiner WordClock Web Portal Lösungen übernehmen? 
Schau mal die Bilder zum integrierten Portal: https://www.printables.com/de/model/768062-wordclock-16x16-2024
Die Integration für Zeitzone und Zeitserver hat mich Anfang des Jahres einige graue Haare gekostet und könnte man für dieses Projekt sicher auch gut integrieren… Dann entfällt das Suchen für die Zeitserver und Zeitzonen Strings die einigen meiner User Kopfzerbrechen bereitet hatten… 

Viele Grüße und schönen Abend.

Blueforcer, 2024-06-03:
Also Ihr=Du, denn ich mach das hier solo ;-)
Das aktuelle Webinterface ist ne thirdparty lib. 
Von Webentwicklung habe ich nicht sooo viel knowhow und vorallem noch weniger Lust xD
Im Grunde ist in AWTRIX ja alles voreingestellt und man muss sich nur die Zeitzone aus der verlinkten liste raussuchen und einfügen.
Aber du bist herzlichst eingeladen das web interface zu überarbeiten xD

AWSW-de, 2024-06-03:
Dann könnten wir uns mal dazu irgendwann mal näher austauschen. Das Portal was ich nutze (ich bastel das auch alles alleine nebenbei) ist auch recht fix aufgesetzt… Halt das gern mal im Hinterkopf und bei Zeiten kann man ja ein paar Ideen zusammen werfen…

Die letzten 2 Jahre hab ich etliche WordClock in nun 14 Sprachen mit je 128 bis 256 LEDs und bis zu 4 Fonts erstellt und das mit den Pixeln so zu lösen ist so viel einfacher und eben dynamisch lösbar….

Bin schon echt froh, Dein Projekt hier gefunden zu haben, denn ich hätte sonst von Vorn mit einem ähnlichen Ansatz neu angefangen, wie oben erwähnt… Das wäre echt schade gewesen, das alles noch mal doppelt zu machen…. 

Nun muss ich nur noch raus finden, wie ich die 0% Akku Anzeige weg bekomme und dann bin ich für heute erst mal sehr zufrieden. 

Darf ich Dein Projekt dann als kompatibel mit meinem 3D Modell benennen auf der oben verlinkten Seite?

PS: Discord ist hinzugefügt. 

Viele Grüße und Danke =)