Blueforcer/awtrix3 #567: [BUG]

jamieyates79, 2024-06-18:
# Bug report

## Describe the bug

Our Ulanzi clocks work reliably for around a week or so, then the screen goes blank. After this happens we can connect to the devices by connecting to their IP addresses, and even see the live view in the browser, but there is no display on the physical device screen. The small light on the back by the charging port shows green. We have two Ulanzi clocks which both experience this issue.

They're both connected to Home Assistant via MQTT and we send occasional messages to them via MQTT. Resetting the clocks resolves the issue for period of time.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96 

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

We expect the display to continue working :)

Blueforcer, 2024-06-18:
Sounds like you're just switching off the matrix :) there is no code related part Woche switches off the matrix randomly. 
 make sure you have no automation which switches off all light entities, because the matrix is a light entity in Homeassistant.
You can turn on the matrix again by double pressing the middle button. (or via the HA entity in Homeassistant or API call)

dom-gunstone, 2024-06-20:
I'm having a similar issue but very, very infrequently. If I had to guess I'd say once every couple of months.

Home assistant shows all of the clock's entities as unavailable until I turn the clock back on by holding the middle button.

Blueforcer, 2024-06-20:
> Home assistant shows all of the clock's entities as unavailable until I turn the clock back on by holding the middle button.

**Holding**  the middle button doesn't turn on or off the clock. It just enter or leave the onscreen Menü.

warp735, 2024-06-21:
I have the same problem. After few days freeze the screen by both Ulanzi´s. Only hard restart with right and left button helps.

dom-gunstone, 2024-06-21:
> **Holding** the middle button doesn't turn on or off the clock. It just enter or leave the onscreen Menü.
> 
> **Double press** turns only the matrix on or off, Awtrix still runs.
> 
> **Holding left and right** switches the power completely on or off.

I probably held the left and right button. It's one of those things that's so infrequent I don't remember the correct way to turn it back on.

Just to be clear, my clock was definitely off. Interestingly, before powering down I saw a decrease in battery level to around 30%. Then when I powered it back on it was about 60% and continued increasing to around 90%. I wonder if I'm having issues with the power supply I'm using. I'll try a different one and see if I have any future issues.

Blueforcer, 2024-06-21:
> Just to be clear, my clock was definitely off. Interestingly, before powering down I saw a decrease in battery level to around 30%. Then when I powered it back on it was about 60% and continued [...]

The software itself isn't able to shut down the clock because there is no physical connection to the power board. Seems like the battery runs out of power.
The battery reading can be inaccurate because there are different clock batches out there with different analog resistors and batteries. You can set you own min and max readings with the dev.json

dom-gunstone, 2024-06-21:
So I probably have a hardware issue. Thanks for the responses.