Blueforcer/awtrix3 #569: [BUG] Can't upload icons

Lavve, 2024-07-04:
# Bug report

## Describe the bug

Loading icons via GUI is no longer possible. In DevTools I get a 500 Internal Server Error when trying to download icon from LaMetrix.
If I via Files try to upload an icon into the ICONS filder I get an alert saying 'ERROR: [500] CREATE FAILED', and DevTools says the same as when trying to download from LaMetrix.

I've tried adding icons both ways in Chrome, Edge and Brave, all three with or without incognito mode and all cache is cleared. Same error everywhere.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior:

Download from LaMetrix:

1. Visit AWTRIX local web interface
2. Click Icons
3. Enter icon ID from LaMetric Icon Gallery, I.E. 54626
4. Click Preview and the icon shows up
5. Clicking the Download button does nothing. Nothing happens. DevTools says '500 Internal Server Error'

Upload via Files

1. Visit AWTRIX local web interface
2. Click Files
3. Choose downloaded 8x8 pixel image
4. Change folder to /ICONS/###.jpg and click Upload
5. A spinner shows up, and then an alert saying 'ERROR: [500] CREATE FAILED'. DevTools says '500 Internal Server Error'

## Expected behavior

When clicking download in the Icons section I expect the icon to be downloaded.
When clicking upload in the Files section I expect the icon to be uiploaded to the ICONS directory

## Screenshots

Screenshot showing the alert from the Files section
[image]

The icon I'm trying to upload
[image]

Don't hesitate to ask me for more information if needed to help me out with this issue.

Blueforcer, 2024-07-04:
How much free disk space do you have? You can see it in the file manager at the top right if you hover over it with the mouse.

Lavve, 2024-07-04:
Haha, that is so funny! Looking at the green (!) bar I thought that it had like 10% filled up, but now when hovering it it says 0.11 MiB free space. Maybe this should be a feature reequest instead then, suggesting showing used space a bit more clearer. And maybe change the bar to red if low space. ;)

Thank you for your help!

Blueforcer, 2024-07-04:
If I remember correctly it will turn red. But of course in a given percentage range. 
So the question is, what's your total space?

Lavve, 2024-07-04:
When hovering it says `1.27 MiB free of 1.38 MiB`.

Blueforcer, 2024-07-04:
Last post you said 0.11 MiB ?!

Lavve, 2024-07-04:
I said 0.11 MiB free space. 1.38 - 1.27 = 0.11

Blueforcer, 2024-07-04:
> I said 0.11 MiB free space. 1.38 - 1.27 = 0.11

That would be the used space.
Like the Filemanager said 
1.27 MiB free of 1.38 MiB

Anyways, it's quite enough for more data.
But since I never heard of this problem, we need to dig a bit deeper.
- I assume you already tried to restart Awtrix3, have you also tried another browser?
- can you manually create a new file with the Filemanager?
- can you download existing icons?

Lavve, 2024-07-04:
Correct, I'm sorry for mixing things up.

I have restarted Awtrix several times, same issue
Creating a file in Filemanager gives me the same error as before; `ERROR: [500] CREATE FAILED`
Yes, I can download files, but now when trying to remove a file the page reloads and the file remains. No error messages. If I try to rename a file, I get `ERROR: [500] RENAME FAILED`

Lavve, 2024-07-10:
Any progress in this issue? Is there anything else I can provide you with to help?

Blueforcer, 2024-07-11:
I'm unable to reproduce it, and after 6k users you're the only one with this problem.
Did you tried to reflash awtrix, with full erase before?

Lavve, 2024-07-12:
That's my last option, I wanted to see if there's a fix for the issue before I start it all over again. I will reflash Awtrix this weekend and come back with the results.

Lavve, 2024-07-13:
Reflashing the Ulanzi with Awtrix again solved the issue, I'm now able to upload icons again.

closed 2024-07-13 by Lavve