Blueforcer/awtrix3 #573: [FEATURE REQUEST] Link to Icon Gallery

Andre1Becker, 2024-07-10:
Hi all,

It would be nice to see a direct link to the [LeMetric Icon Gallery](https://developer.lametric.com/icons) in the docs or even better, within the "Icons" tab on the Webinterface.

Cheers!

daniel-simpson, 2024-09-03:
You can actually do this yourself, if you're willing to play around in the `DoNotTouch.json` file.

Note: back it up first!

Change the line starting with `  "raw-html-icon_html":` to

```json
"raw-html-icon_html": " <div id=\"iconcontent\"> <div id=\"form-con\"> <form> <label for=\"lametric-iconID\">Icon ID</label><br> <input type=\"text\" id=\"lametric-iconID\" name=\"lametric-iconID\"><br> <div class=\"button-row\"> <input class=\"btn\" type=\"button\" value=\"Preview\" onclick=\"createLametricLink()\"> <input class=\"btn\" type=\"button\" value=\"Download\" onclick=\"downloadLametricImage()\"> </div> </form> </div> <br> <br> <div id=\"icon-container\"> </div> <a href=\"https://developer.lametric.com/icons\" target=\"_blank\">Icon lists</a> </div>",
```

Not particularly pretty, but gets the job done!

[image]

Andre1Becker, 2024-09-03:
Works as expected! Nice!

closed 2024-09-03 by Andre1Becker