Blueforcer/awtrix3 #576: [BUG] Strange icon color behavior of animated icon

UCaGk, 2024-07-19:
I created different icons/animated icons an LaMetric side.
Some of the animated icons have a strange behavior. 
One animation is a door with yellow frame, first line of icons.
When the red exclamation mark removed the white field inside the yello frame gets red ? (Line 2 of icons)
The other animation is a door, third lin of icons.
When the green check mark removed the hole yellow frame gets green ? (Line 2 of icons)

[image]

Blueforcer, 2024-07-19:
Not every GIF is compatible. (See documentation)

You can try to strip down the color table 
https://github.com/Blueforcer/awtrix3/issues/560#issuecomment-2169141158

Or use the icon creator of the awtrix app. This will always work.

UCaGk, 2024-07-19:
Ah, okay, thanks for your help !

closed 2024-08-15 by Blueforcer