Blueforcer/awtrix3 #579: [BUG] VLANs supported?

canoben, 2024-07-22:
# Bug report

## Describe the bug

I configured my PixelClock initially for usage with a wireless network associated to a VLAN used by all of my IoT devices. Although I could see (and configure) the device just fine using the browser on my main network, the Home Assistant instance running on a server on the main network could not see the device.
Putting the device on the main network and it is immediately recognised by HA. All my VLANs have mDNS activated.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

## To Reproduce

put device on IoT VLAN -> not connection to HA
put device on main VLAN -> works fine

## Expected behavior

With mDNS activated, device should be seen on any network HA is running

Anybody else saw this behaviour, and if so, knows a fix?

Blueforcer, 2024-07-22:
Not sure wheres the bug, if it's a network problem. 
HA Discovery works the other way. 
AWTRIX sends a couple of Mqtt messages to the HomeAssistant broker, so HA knows which entities to create.
So make sure awtrix can communicate with the mqtt broker on your main vlan.

canoben, 2024-07-22:
My understanding of mDNS may very well be wrong, but this pops up in my router software when hovering over the mDNS option:

[image]

Anyway, I created a firewall rule granting access to the Pixel Clock from the IoT VLAN to the Home Assistant instance IP address (and only that one) and it is working fine now.

Blueforcer, 2024-07-22:
mDNS has nothing to do with the HA Discovery. It's a complete own proprietary "protocol" by HomeAssistant

closed 2024-09-12 by Blueforcer