Blueforcer/awtrix3 #580: [BUG] Awetrix turns off sometimes

BenAhrdt, 2024-07-24:
# Bug report
The device turns off the power.

## Describe the bug
Sometimes in the morning i see, that the device is powered off.
press the left and right button tuns it on. the device is continously on usb supply.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

Blueforcer, 2024-07-24:
Then it's not enough power. The software is not able to turn off the device. There is no physical connection to the powerboard

BenAhrdt, 2024-07-24:
But it is still on the usb an when i turn it on, it works without usb. => so the battery is full

Blueforcer, 2024-07-24:
It's a hardware related issue, try another USB source, make sure you doesn't switch off the power supply over night etc.
 It's 100% impossible that the firmware can turn off the device.

BenAhrdt, 2024-07-24:
OK. i will try.

mod42, 2024-08-10:
I see the same. After some time, the clock switches off. Always connected via USB.

ploro-it, 2024-09-05:
I have the same issue: when the clock is under 70% battery, it suddenly goes off, and I have to turn it on pressing the two buttons. But it only lasts a few minutes, until I recharge it to 100%. I don't think it is a software related bug, because I had this problem with both the old firmware and the new one

Blueforcer, 2024-09-05:
There are different batches of the Ulanzi with different battery readings. 
Use the raw_bat value of the stats and set the min and max battery values in the dev.json to make the percentage display correct

closed 2024-09-12 by Blueforcer