Blueforcer/awtrix3 #581: [BUG] Day Of Week Color

FrankGiesecke, 2024-07-24:
# Bug report

## Describe the bug

If I understand the API docu, it should be possible to define a color for day of week marker using the keys "WDCA" and "WDCI" in the settings endpoint.

Unfortunately this has no effect.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

## To Reproduce

Send the following settings config in MQTT

```json
{
"WDCA": "#FF0000",
"WDCI": "#0000FF"
}
```

## Expected behavior

The current day should be represented in red, all other days in blue.

## Screenshots

-

## Logs

-

## Additional context

-

EmEiBee, 2024-10-28:
I agree with that.
Almost all other commands in config also have no effect, such as SOM or DAT

Blueforcer, 2024-10-28:
Works as it should.

```json
{
"WDCA": "#FF0000",
"WDCI": "#0000FF",
"SOM":false
}```
results in the correct behaviour:

![awtrix (2)](https://github.com/user-attachments/assets/fbfeaf66-00ce-42e6-843b-15f4e16fd762)

EmEiBee, 2024-10-28:
... ... unfortunately not for me:
[image]

Blueforcer, 2024-10-28:
> ... ... unfortunately not for me: 

its /settings not /settings/

thats why the messagetree looks different to mine:
[image]

EmEiBee, 2024-10-28:
> its /settings not /settings/

UFFF... Ok, you are right here... Then it works like expected...

So Q:
Why I have to use it without backslash at this point where others have to be with backslash?

[image]

Blueforcer, 2024-10-28:
This is also not the case here
Your Topic is like pixel/custom/HKV not pixel/custom/

closed 2024-11-14 by Blueforcer