Blueforcer/awtrix3 #582: [BUG] can not access my wifi ssid connection error

goofy84, 2024-07-29:
## Describe the bug
i flashed the firmware, ulanzi watch is in ap mode, accessing the watch with my celluar phone. Scanning ssid is working, typed in the correct password. The result is always connection error. Maybe the ssid is the problem, i have a space in the string: FRITZBox 24

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ v0.96 ]

FritzTheBlitz, 2024-09-07:
Same config and device. Two spaces in SID. No problem.

Sigi-Bereuter, 2025-02-05:
Same here: 
1. After Unpacking the TC001 I was able to access the WebApp via 192.168.4.1, can scan Wifi Networks and shows my SSID.
I was not able to connect to multiple SSID's with correct pwd.

2. Flashed Awtrix 0.98 
Again can scan Wifi Networks but with correct Pwd no connection to SSID.
Display stay constantly in "AP Mode" even if I press "Connect SSID" in the WebApp

Log:

[2040] [startLittleFS]: Starting filesystem
[2059] [startLittleFS]: Filesystem started
[2060] [loadSettings]: Loading Usersettings
[2062] [getID]: Starting filesystem
[2062] [loadDevSettings]: Loading Devsettings
[2074] [loadDevSettings]: Devsettings not found
[2075] [setup]: Setup periphery
[2089] [setup]: SHT31 sensor detected
[2091] [loadSettings]: Webserver configuration file not exist
[2092] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to Moria
...................................................No connection after a while -> go in Access Point mode

AP mode.
Server IP address: 192.168.4.1

[18067] [setup]: My IP: 192.168.4.1
[2040] [startLittleFS]: Starting filesystem
[2059] [startLittleFS]: Filesystem started
[2060] [loadSettings]: Loading Usersettings
[2062] [getID]: Starting filesystem
[2062] [loadDevSettings]: Loading Devsettings
[2074] [loadDevSettings]: Devsettings not found
[2075] [setup]: Setup periphery
[2089] [setup]: SHT31 sensor detected
[2091] [loadSettings]: Webserver configuration file not exist
[2092] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to Moria
...................................................No connection after a while -> go in Access Point mode

AP mode.
Server IP address: 192.168.4.1

[18066] [setup]: My IP: 192.168.4.1
>