Blueforcer/awtrix3 #583: [FEATURE REQUEST] Send an MQTT message from the device when a text string has finished scrolling.

gaussey, 2024-07-29:
# Feature Request

### Is your feature request related to a problem?

If so, then add a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution / feature you'd like

A clear and concise description of what you want to happen.

### Describe alternatives you've considered

A clear and concise description of any alternative solutions or features you've considered.

### Additional context

Add any other context or screenshots about the feature request here.

I'm using the clock (or will be) as much nicer replacement for my MAX7219 matrix that shows the contest of RSS feeds and data scraped from webpages using HomeAssistant as the backend and MQTT broker. Currently with my dumb MAX7219 I calculate when a feed will have finished based on its character length and then send a new MQTT message -  Awtrix doesn't allow you to stop a message midscroll that I can tell so I have to set duration to the correct length programatically instead.. It'd be nice if the Awtrix flashed device could send an MQTT message when it had finished scrolling so I could use that as a trigger to send the next stream of text data. If something like this is already available I'd love to know about it.

Blueforcer, 2024-07-29:
I don't understand the problem.
Just use “repeat” instead of “duration” and awtrix automatically scrolls the whole text the complete length, and then switches to the next app. so you can also just listen to currentApp and when this switches from your app to the other one you send new data.

gaussey, 2024-07-29:
But I'm not switching apps - I disabled all the built in apps, I'm just sending multiple MQTT messages one after another with text strings.

Blueforcer, 2024-07-29:
I do not assume that such a special feast reuquest will be implemented for this case in the near future.
You could create another feed app and have it alternate and always update the app that is not shown.

gaussey, 2024-07-29:
Yeah, that's an idea. I was kind of hoping I was missing something obvious but thank you for responding.

closed 2025-03-03 by Blueforcer