Blueforcer/awtrix3 #584: [BUG] Icon Tab missing

bubiman, 2024-07-30:
# Bug report

## Describe the bug

The Icon Tab to download and install icon is missing. 
Yesterday I received my TC001 flashed it with version 0.96 and it was working fine. I could download icons yesterday but today the Icon Tab is simply no longer avaible. I tried to reboot the clock through iobroker, I turn the clock off and back on again the the arrow buttons but the Icon Tab is still missing

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

## Screenshots

[image]

## Additional context

I can see and use the icons I have installed yesterday

[image]

udo1toni, 2024-07-31:
Maybe you messed up the DoNotTouch.json.
Afaik deleting the file and do a restart should create a new (default) file. Please be aware that you will lose some settings such as ntp server or a fixed ip.

bubiman, 2024-08-01:
I haven`t touched the DoNotTouch.json. I have reset the clock with the button on the back and had to set up the clock again. For now it is working

eku, 2024-08-02:
> Maybe you messed up the DoNotTouch.json.

The software itself is responsible for this, see also #549.

closed 2024-08-15 by Blueforcer