Blueforcer/awtrix3 #589: [BUG] RTTTL pause does not work

baxerus, 2024-08-12:
# Bug report

## Describe the bug

When you use a RTTTL sound with one (or multiple) **pause** (`p`) in it, there is no silence when the pause sign is played, but instead the last tone is played *longer* for the length of the pause

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96

## To Reproduce

Steps to reproduce the behavior:

Send a notification via `MQTT` to the Awtrix with the the following test sound in it:
`s:d=4,o=5,b=250:d,p,p,p,p,p,p,p,f`.

Compare the output played by the Awtrix with what https://adamonsoon.github.io/rtttl-play/ and https://corax89.github.io/esp8266Game/soundEditor.html plays for this wonderful example `s:d=4,o=5,b=250:d,p,p,p,p,p,p,p,f`.

## Expected behavior

When there is a pause (`p`), no tone is played.

baxerus, 2024-08-12:
Yet?

Between my bug report and your message are around 4 minutes 😃

Blueforcer, 2024-08-12:
I know this problem for a while. Buts a external library and I don't understand where the problem is.

baxerus, 2024-08-12:
My C++ skills are bad.
I glanced through the code here on github a bit. But I didn't even find out, where the `noTone()` function is defined (until there everything at least looked okay 😉)

baxerus, 2024-08-12:
Also looked a bit through the code of the library.
[This](https://github.com/fabianoriccardi/melody-player/blame/fd7a04492b9aeb690b6f446182d08eeb86880fe1/src/melody_player.cpp#L225) is the first occurrence of  `noTone()` I could find...

In general the string `noTone` is only two times in the code.

My assumption is, it was **never** implemented and was simply forgotten 😃
If this is even possible without the code crashing (here my C++ knowledge about error handling or not handling is not enough)...

closed 2025-03-03 by Blueforcer

baxerus, 2025-03-04:
Does `completed` means it is fixed?