Blueforcer/awtrix3 #590: [INFO] Is it possible to update firmware ?

usky73, 2024-08-14:
I have a TC001, I did the firmware update few month ago.
My firmware is AWTRIX Light 0.70, it seems a little bit old. 
The link on web interface to the documentation is not working anymore.
I remember than flashing the firmware was a nightmare (might be due to the USB cable) I would like to avoid it again, and more than that, not loosing the setting.

I have an update menu on web interface

Is there a way to use it to update it ?
Where can I find the update file ?
To save config and import it on awtrix3 ?

klein0r, 2024-08-15:
> The link on web interface to the documentation is not working anymore.

The project has been renamed to Awtrix 3 and the urls have changed. The web flasher is available here:

https://blueforcer.github.io/awtrix3/#/flasher

Blueforcer, 2024-08-15:
Where did you find the old link? I thought I had replaced them all.

usky73, 2024-08-15:
The link came from the web interface of the firmware - AWTRIX Light 0.70

usky73, 2024-08-15:
> > The link on web interface to the documentation is not working anymore.
> 
> The project has been renamed to Awtrix 3 and the urls have changed. The web flasher is available here:
> 
> https://blueforcer.github.io/awtrix3/#/flasher

Then no other choice than flash it again, correct ?

Blueforcer, 2024-08-15:
You can download the bin file from the repo releases and upload it in the update Menü. But if it's too old you need to reflash it

usky73, 2024-08-15:
ok thanks.

usky73, 2024-08-15:
last dummy question, where is this bin file ?

Blueforcer, 2024-08-15:
https://github.com/Blueforcer/awtrix3/releases

closed 2024-08-16 by Blueforcer