Blueforcer/awtrix3 #591: [BUG]

FrankGiesecke, 2024-08-15:
# Bug report

## Describe the bug

I'd like to change the duration an app is displayed. I used Home Assistant to generate a MQTT message to modify the attribute "ATIME", but it doesn't change anithing. The duration is still 7 seconds.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior:

```yaml
domain: mqtt
service: publish
service_data:
  qos: 0
  retain: false
  topic: clock01/settings
  payload: |-
    {
      "ATIME": 3,
      "TIM": True,
      "TFORMAT": "%H:%M:%S",
      "TMODE": 0,
      "DAT": True,
      "SOM": true,
      "HUM": False,
      "TEMP": False,
      "BAT": False
    }
target: {}
```

## Expected behavior

An app should only be displayed for 3 seconds.

Blueforcer, 2024-08-15:
Your json is incorrect. Boolean values must be written in lower case.

FrankGiesecke, 2024-08-15:
> Your json is incorrect and cant be parsed correctly. Boolean values must be written in lower case.

Oh, I missed this because it was generated.
Fixing this did the trick. Thanks very much :)

closed 2024-08-15 by FrankGiesecke