Blueforcer/awtrix3 #593: [BUG] Ulanzi TC0001/Awtrix 3 doesnt respond to api calls anymore

BitWire, 2024-08-22:
# Bug report

## Describe the bug

Ive run v0.96 since release, but now for like a couple weeks, even tho it connects to wifi, it doesnt pull the correct time or reacts to api calls to its API, the requests just time out. It also just did that out of the blue.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ v0.96 ]

## To Reproduce

Keep your Ulanzi on for months and only turn off the display, after a couple months this will happen, and will be consistently happening even after proper full restarts or reflashes.

## Expected behavior

The Ulanzi should keep reacting to api calls.

## Logs

I cant really pull logs because even if i connect it via usb it doesnt want to talk to the flasher in chrome.

BitWire, 2024-08-22:
A couple hours after i wrote this ticket, the Clock suddenly decided to just reboot and is stuck in AP Mode without being able to connect to it, as it hasnt created a wifi network.

c1328, 2024-08-24:
...same here. Not the reboot into AP mode yet, but it more and more often does not respond to API calls and I have manually reboot the device.

jasonrichardsmith, 2025-01-17:
Same issue here.  I performed a manual firmware update (0.98) with the clocks web interface.  I have two clocks, this morning one is responding the other is not.  They are about 10 feet away from each other. I am going to set them up to restart every three hours and see if it resolves the issue.

jasonrichardsmith, 2025-01-26:
I was able to solve the problem, by putting them closer to the antenna, and putting them on a wifi that only supports 2.4ghz not 5 and 2.4 (which is the default in the house).  No problem since.