Blueforcer/awtrix3 #594: [BUG] Clocks don't connect to Wifi after battery runs out

ThaMighty90, 2024-08-25:
# Bug report

## Describe the bug

My clocks are powered by a power brick that gets turned off during times I am not at my PC. So sometimes they run completly out off battery. After rebooting them they do not properly connect to my wifi. when I then join there network to configure SSID and Password I can see that they still know where to connect to they just don't do it. I have to reenter my credentials and then they reconnect but not always sometimes I have to reboot again and enter my credentials again.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior:
Let clock run out of battery recharge and boot up the clock

## Expected behavior

They just connect to their configured network

AxeWax, 2024-10-13:
I am having the same Problem. The IP is still correct, SSID and Wifi PW did not change. But its not reconnecting to Wifi.
The only thing that changed, in my case, was my new Wifi Access Point.

Blueforcer, 2024-11-14:
Sounds more like the connection quality is just not good enough, so it's more a matter of luck that it can connect. please test it if you put the awtrix closer to the wifi AP.

closed 2024-12-23 by Blueforcer