Blueforcer/awtrix3 #595: [FEATURE REQUEST] Offer an option to calibrate the reported illuminance value

vlesquin, 2024-08-30:
# Feature Request

I'm using the reported illuminance value (in lx) to switch automatically with Homeassistant to a night mode the Unlanzi device under AWTRIX.

However, the reported value is too low, in the range of 0-50lux but reaching too easily 0lx in my case, resulting to a night mode during the day.

In order to have an efficient automation, I would like to have the option to boost the reporter value by a given factor to better calibrate the behaviour.

thanks for your help !

Blueforcer, 2024-08-30:
since nobody knows which sensor is build in, it's not possible to calibrate it.
But for automation you can get the raw value from the stats topic and key "ldr_raw"

vlesquin, 2024-08-30:
Thanks a lot Stephan for a such quick answer and the tip !

closed 2024-08-30 by vlesquin