Blueforcer/awtrix3 #597: [BUG] Auth breaks Icon upload

EricReiche, 2024-09-05:
# Bug report

## Describe the bug
I tried to enable auth (user+pw) for my device, but it seems that breaks the icon upload. If I try to do it from the webinterface, it simply doesn't do anything after clicking the download button. When using an external site like https://flows.blueforcer.de/ it crashes the whole device and I have to reset it via Hardware buttons. After disabling auth (which for some reason I could only do on Firefox, Chrome refused to remove the user+pw), it works again.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce
- Enable user & password.
- Restart device
- Try to upload an icon, e.g. 54156

## Expected behavior
Dialog "Finish - Icon saved" should open

deurk, 2024-09-13:
Downloading icons from LaMetric website through the UI does not work with Awtrix auth configured. Previewing is okay but clicking the Download button will end up in `ID cannot be found` message.

deurk, 2024-09-16:
> Downloading icons from LaMetric website through the UI does not work with Awtrix auth configured. Previewing is okay but clicking the Download button will end up in `ID cannot be found` message.

Forget what I said, in my case it was an ad blocker that was not allowing the web proxy used by awtrix3 to download the icons.