Blueforcer/awtrix3 #599: [BUG]

FritzTheBlitz, 2024-09-07:
# Bug report

## Describe the bug

Playing around with the TEFF JSON property and setting the value to 3 (Zoom) via http, when the next animation starts, the device turns off. When turning it back on, after the IP is showed, again the animation tries to start, and the device turns off again.
I had to flash the device again (you have to be quick), what pleasingly simple is. Thank you!

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior (**be carful**):
`curl -X POST --data "{'TEFF':'3'}" -H "Content-Type: application/json" http://device-ip/api/settings`

## Expected behavior

The device should animate the transition with the zoom animation.

Blueforcer, 2024-09-12:
According to the documentation the value needs to be an number not a string.
So it should be `{'TEFF':3}`
Currently there is no check for the given object type.

FritzTheBlitz, 2024-09-12:
Thank you! It works as described. Good work by the way

closed 2024-09-12 by FritzTheBlitz