Blueforcer/awtrix3 #601: [BUG]

DeDust1337, 2024-09-13:
# Bug report

## Describe the bug

After crashing AWTRIX suddenly changed its own IP. Now it doesnt connect to WIFI after entering SSID and password. Several tabs are not displayed (as mqtt). Flashing to another/the same version doesnt show any effect. Trying to get back to factory firmware doesnt work either.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96 & v0.94

## To Reproduce

No idea. It happened out of nowhere.

## Expected behavior

To properly connect to wifi / being able to at least reset it.

## Screenshots

[image]

## Logs

When trying to flash it with the download tool cmd displays: 

[ESP8266Loader_spi[1]][espDownloader.py][line:766][ERROR]:
no log file output ...

Blueforcer, 2024-09-13:
Try deleting the doNotTouch.json and reboot awtrix. You would need to enter time and mqtt settings in the Webinterface again

DeDust1337, 2024-09-13:
Danke ich versuchs Mal

DeDust1337, 2024-09-13:
Just deleting the doNotTouch.json did actually work. Thanks and keep up the great work!

closed 2024-09-13 by DeDust1337