Blueforcer/awtrix3 #604: [BUG] Stuck in AP Mode

2K-Darkness, 2024-09-29:
# Bug report

## Describe the bug

From one day of another, my ulanzi wont show any Updates. Meaning, it was still recognizing my buttons and Switch es trough all configured sites, but it wont refresh new values. No Wifi connection issue with a Red dot or things like These ...
I restarted the device and got stuck in AP Mode...
I connectrd to the clock and was able to scan my wifi but it is Not possible to connect. It says error or Password too Long!?
My PW is 16 chars...

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96

## To Reproduce

Steps to reproduce the behavior:
Dont know if its only in my device!?

## Expected behavior

Connect succesfully to my wifi again

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

udo1toni, 2024-09-29:
I'm using over 40 characters for password, so I doubt that this is causing any problem. Maybe try to reflash the device?

2K-Darkness, 2024-09-29:
Understand...
I also doubt that, but i noticed a bugfix in a newer release, where storing of Long ssid or PW was an issue.
However, is downgrade possible? Or is maybe a reset?
I already tried an overflashing to 96 alteady

2K-Darkness, 2024-09-30:
Nope... Ok!

In the meanwhile i found the issue.
There were two identical SSIDs in my WiFi Area. One of them did not have Internet anymore.
However, i tested both SSIDs (as both were given in GUI after Scanning with the same Naming) and none of them worked.
But anyway, i found the wrong one and disabled that AP. After that, Connection was successfull again!

Thanks for your Help!

Regards

closed 2024-12-23 by Blueforcer