Blueforcer/awtrix3 #605: [FEATURE REQUEST] Always on mode

lchojnowski, 2024-10-01:
My Ulanzi TC001 AWTRIX is attached to a power source which is always switched off during night and switched on once I arrive to the office. It's frustrating that I need to switch clock manually every day. What we need is ability to set always on mode, which mean the AWTRIX is always on if there is power supply on.

Blueforcer, 2024-10-01:
It's not possible, if the battery is empty the power board switches off the ESP32.
If you hold both buttons, you doesn't give a signal to the ESP, but to the power board, to turn on the ESP.

closed 2024-10-01 by Blueforcer

lchojnowski, 2024-10-02:
For anyone looking for that: https://www.youtube.com/watch?v=HIQl6D0DVNE - here's the hack (you need to remove battery).