Blueforcer/awtrix3 #611: [BUG] Stuck in AP, click "eye" for showing PWD popup RESET

EmEiBee, 2024-10-11:
Received a brand new TC001 today.
After connecting to the PC and flashing in EDGE _WITH_ erase (twice), the TC now shows "AP MODE".
After establishing a connection via WLAN, I can call up the GUI at 192.168.4.1 to enter my WLAN data.

Problem 1:
If I click on the eye to display the password in plain text, a window pops up with the message "RESTART! - ESP restarted. Please wait a little and then reload this page."

Problem 2:
Although the entries for the local WLAN are correct, the TC does not want to establish a connection to the WLAN (IPv6 is switched off in the WLAN, 2.4GHz and 5GHz have the same SSID, which does not cause any problems with other ESPs (EasyESP, Tasmota))

- Ulanzi TC001 SKU 2882
- awtrix3 version: 0v96

LOG:
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256

AP mode.
```
Server IP address: 192.168.4.1

Connecting to BUCHHOLZ
*.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.**.*Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to BUCHHOLZ
..................................................No connection after a while -> go in Access Point mode
E (17911) wifi:sta is connecting, return error

AP mode.
Server IP address: 192.168.4.1
```

[image]

I also have connected hTerm. Maybe to see more what's going on... nope...

If the problem cannot be solved this way, would it be possible to use the terminal (hTerm) to provide the device with the login data, as is possible with EasyESP, for example?

udo1toni, 2024-10-12:
Maybe some exotic chars in the password?

EmEiBee, 2024-10-12:
Nope... Only letters without special characters.
As I said: All other ESP based devices (mainly Shelly, and DIY based on 8266 and x32 with EasyESP or Tasmota) work so far in the same WLAN network with the same SSID and same PWD

EmEiBee, 2024-10-13:
Update:
My wife was at a flea market yesterday. As luck would have it, a woman was offering a TC001. Used, not flashed, a bit dusty, but otherwise OK.
Flashing was absolutely no problem here (even though I'm not a fan of flashing via the browser).
So I suspect that the ESP on the first TC001, although brand new, has a problem. I'll put it aside for now and replace it when I get the chance.

So it's **not** a BUG, ​​it's just disguised as one.

closed 2024-10-13 by EmEiBee