Blueforcer/awtrix3 #613: [BUG] Auto poweroff

jissoe, 2024-10-13:
The clock shuts down after some hours with turned off matrix. I have to start it by pressing the two buttons and disable the default apps again... 

I don't think there battery runs empty, it has over 50% when I start the clock again (Without power input).
It only gets charged when my TV is on, but the matrix is turned off in this time.

Blueforcer, 2024-10-13:
Awtrix can't shutdown itself, there is no connection between the ESP and power board.

But since there are different batches with different battery measurements out there, it could be possible that you have to set the min/max values for the battery readings by yourself in the dev.josn.

jissoe, 2024-10-13:
So it's the hardware which shuts the clock off? I mean there must be something shutting down, because the battery does not seem to be drained. I will test how long it lasts with 50% to see if the voltage has to be calibrated. 

Is it possible to disable default apps somehow? With mqtt commands I have to restart the clock

Blueforcer, 2024-10-13:
in your case 50% seems to be empty. There are already users wich has the same problem.

I don't know the runtime, I removed the battery.

jissoe, 2024-10-13:
[image]

It lasts about 8h30, seems too short for me
Is there anything to reduce consumption besides turning the matrix off?

Blueforcer, 2024-10-13:
Here are values from a discord user:

{
  "min_battery":565,
  "max_battery":656
}

please note, the hardware is pretty cheap and also the battery. It won't last long and the capacity will be reduced pretty fast over time.

Blueforcer, 2024-10-13:
I must have deleted your post by mistake.

8.5h is completely ok for the small battery. There is nothing you can do.

Petro31, 2024-11-02:
I'm having a similar issue it seems.  Just got the device, device just shuts down on or off power supply.  Battery appears to be around 50% when this happens.

What is the recommended power supply for these things?  I don't plan on running it off battery.

Petro31, 2024-11-02:
The clock is now dying exactly 10 minutes from boot.

Blueforcer, 2024-11-02:
> The clock is now shutting down exactly 10 minutes from boot.

Probably because the battery is empty.
Turn it off. Charge it with a stronger power supply. And set the measured limits with the dev.json

Rustymage, 2024-11-03:
I removed my battery and my clock now randomly turns off and requires rebooting :/

Petro31, 2024-11-05:
> > The clock is now shutting down exactly 10 minutes from boot.
> 
> Probably because the battery is empty. Turn it off. Charge it with a stronger power supply. And set the measured limits with the dev.json

Was the power supply being 5v 1a, which is not enough to charge the battery.  5v 2a seems to work well.

Sonsi79, 2024-11-09:
Same for me. My clock turned off also when it is plugged in on a charger.

jissoe, 2024-11-09:
Mine is running without problems since I don't switch off the powersupply.