Blueforcer/awtrix3 #616: MQTT: topText / hold dont work?

EmEiBee, 2024-10-16:
Hello there,

could I get some help please?
I send the following to the watch (awtrix 0v96) via MQTT (Explorer):

```
{
"text":"HEATING FAULT!",
"color":"#FF0000",
"topText":true,
"hold":true
};

```
I assumed that -topText- would display the message immediately and suppress all apps currently running, and that -hold- would display it until it was acknowledged or the "alarm" app was deleted.
The text is displayed in red, but only at the end of all running apps (time, date, outside temperature, outside humidity).

Am I misunderstanding something here???

Blueforcer, 2024-10-16:
Toptext only belongs to the current data, so the text will be shown over a icon or drawing.

I assume you're sending it to the /custom API. Send it to /notify. This is a notification and will be displayed as a top layer over everything.

EmEiBee, 2024-10-16:
Ohhhh ^^ Yes, that makes sense now that you mention it...
Thank you for this tip! I somehow didn't have it on my radar...

Greetz
Micha

Blueforcer, 2024-10-16:
Der MQTT explorer zeigt dir wo der Fehler ist. Das Semikolon am Ende muss weg. Sonst ist die Json nicht valide

Blueforcer, 2024-10-16:
https://blueforcer.github.io/awtrix3/#/api?id=dismiss-notification

EmEiBee, 2024-10-16:
Danke! Auf die Idee nach dem Wort "dismiss" zu suchen, bin ich gar nicht gekommen. 
Sollte man vielleicht mal mit in die Beschreibung von "hold" mit reinnehmen...

EmEiBee, 2024-10-16:
"loopSound" funktioniert weder mit der direkten Übergabe von rtttl-Daten, noch mit unter /Melodies abgelegten und via "sound" aufgerufenen rtttl-Dateien.

EmEiBee, 2024-10-17:
hmmm...
Ist das nun ein Bug oder bin ich zu blöd?

EmEiBee, 2024-11-14:
Dann verstehe ich es nicht... 
Ich setze mal ....
`/pixel/notify {"sound":"rtttl_Sirene","loopSound":true}`
An der Stelle würde ich davon ausgehen, dass hier die Loop, ähnlich wie 'hold', so lange dudelt, bis das notify quittiert wird. Das passiert nicht. Zudem löscht sich das notify selbst. Es läuft nur so lange, wie die Anzeigezeit der anderen App's gesetzt ist. Ist diese 5 Sekunden, dann dudelt es auch 5 Sekunden, gefolgt von der ersten App in der Reihe und zudem am Ende nicht mehr vorhanden.
Es funktioniert also nur so:
`/pixel/notify {"sound":"rtttl_Sirene","loopSound":true,"hold":true}`
Das ist nach meiner, natürlich unmaßgeblichen, Meinung irgendwie doppelt gemoppelt. Oder verstehe ich hier irgendwas falsch?

Blueforcer, 2024-11-14:
nö,
warum sollte der notification sound weiter loopen, wenn die notification beendet ist?
So hat doch jeder die Möglichkeit es so anzupassen wie es ihm gefällt. 
Ewige notification mit ewigem gedudel, oder ewige notification mit einmal gedudel.

EmEiBee, 2024-11-14:
... hmmmm ... Von der Seite habe ich das noch nicht betrachtet; da magst Du recht haben...
Ok, ist so dann verstanden worden. Danke dafür! Habe ich mir gleich mal als Notiz dran gehängt, damit ich es nächstes Jahr auch noch weiß ;)

EDIT:
Es wäre schön, wenn man bei LoopSound noch eine Pausenzeit angeben könnte, bestenfalls anstatt 'true' einen Wert größer 0 in z.B. 'Pause  * 100ms'. Man könnte es sich zwar in den rtttl-sound mit einbauen am Ende, aber dann bräuchte man oft von einem Sound gleich zwei oder mehr Versionen mit unterschiedlichen Pausen am Ende (funktioniert das überhaupt? Noch nicht probiert); auch nicht schön...

Blueforcer, 2024-11-14:
Ich notiers mal, denke aber das es nicht so einfach umzusetzen ist.
Ich mach hier mal dicht, da sich das Ursprungs issue ja schon länger erledigt hat

closed 2024-11-14 by Blueforcer