Blueforcer/awtrix3 #617: [BUG] icon tab in web interface didn't have LaMetric integration and icon id assignment

veerakri555, 2024-10-19:
I just bought Ulanzi TC001 and installed AWTRIX3.  I also bought AWTRIX3 android mobile app.   I can download the icon to the clock but when I go to icon tab, it only has Icon ID input box with preview and download buttons.  There is no LaMatric input box or place where can assign ID to the downloaded icon.  Please see the attached screen capture.

[image]

Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

veerakri555, 2024-10-19:
What I can do right now is to type the full name of the icon instead of icon ID.  For example, i downloaded Temperatur2 icon from Awtrix 3 mobile app.  My mqtt payload would be     { "text" : "{{ states('sensor.temperature_average_formatted') }}", 
      "icon" : "Temperatur2", "rainbow" : true, "duration" : 10 }

Blueforcer, 2024-10-19:
What do you mean? Icon IDs are only for lametric icons.
If you're using awtrix 3 icons you can use the name of the file.

veerakri555, 2024-10-19:
This recent youtube explain the way to get the LaMatric icon.  It seems like this version change the flow of how to get LaMatric icon.  https://www.youtube.com/watch?v=jQy8D_eXoKY&t=91s

closed 2024-10-19 by veerakri555