Blueforcer/awtrix3 #618: AWTRIX does not render integers or numeric sensor data in YAML templates

Turtlepaw, 2024-10-20:
# Bug report

## Describe the bug

AWTRIX fails to render integer values (such as sensor data) when the output consists solely of a number. However, when appending a non-numeric character (e.g., a space, letter, or symbol), AWTRIX renders the value correctly.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

Steps to reproduce the behavior:

- Set up AWTRIX with a custom sensor app to display step count or other numeric values (e.g., from a Home Assistant sensor).
- Use a YAML template that outputs an integer value, such as:
```yaml
text: >
  {% set steps = states('sensor.google_pixel_watch_daily_steps_4') | int %}
  {{ steps }}
```
- AWTRIX fails to display anything (blank screen).
- Modify the template by appending a non-numeric character (e.g., a space, letter, or symbol) to the output, such as:
```yaml
text: >
  {% set steps = states('sensor.google_pixel_watch_daily_steps_4') | int %}
  {{ steps ~ '*' }}
```
- AWTRIX now renders the text (e.g., 5000*).

## Expected behavior

AWTRIX should render the integer (e.g., `5000`) directly without requiring any additional characters.

## Screenshots

[image]

## Logs

No logs

## Workaround
Appending an invisible character (like a zero-width space) or a visible symbol to the output forces AWTRIX to display the numeric value.

### Example working code:

```yaml
text: >
  {% set steps = states('sensor.google_pixel_watch_daily_steps_4') | int %}
  {{ steps ~ '​' }}  # Appending zero-width space
```

## Additional context

This bug seems related to how AWTRIX processes or parses numeric values in YAML templates. It treats numeric-only outputs as invalid and doesn’t render them unless a non-numeric character is appended.

eku, 2024-10-20:
What does the JSON structure that is sent to Awtrix look like?

Blueforcer, 2024-10-20:
According to the documentation, you need to send a String, not a number. 

So it should be

`text: > 
  {% set steps = states('sensor.google_pixel_watch_daily_steps_4') %} 
  {{ steps }}`

Turtlepaw, 2024-10-21:
When using the code you provided:

> ```yaml
> text: >
>   {% set steps = states('sensor.google_pixel_watch_daily_steps_4') | string %}
>   {{ steps }}
> ```

It still fails to render the text.

<details>
  <summary>📸 Awtrix Image</summary>
  <p>

[image]
</p>
</details>

<details>
  <summary>🤖 Home Assistant Automation</summary>
  <p>

This automation uses a third-party library: [10der/homeassistant-custom_components-awtrix](https://github.com/10der/homeassistant-custom_components-awtrix)

```yaml
alias: Awtrix Steps
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.google_pixel_watch_daily_steps_4
  - trigger: time_pattern
    minutes: "14"
conditions: []
actions:
  - action: awtrix.awtrix_2d8db4_push_app_data
    data:
      name: Steps
      data:
        text: >
          {% set steps = states('sensor.google_pixel_watch_daily_steps_4') |
          string %}  {{ steps }}
        icon: "30415"
        duration: 0
        pushIcon: 0
        lifetime: 900
        repeat: 1
mode: single
```

</p>
</details>

Honusnap, 2024-10-24:
The problem is most certainly caused by the way the text is sent to the publish order, it's lost somewhere.

daath, 2024-12-30:
Same issue here - clearly a bug (as casting to a string does not work either).

Blueforcer, 2024-12-30:
> Same issue here - clearly a bug (as casting to a string does not work either).

Its not a bug. the key "text" needs to be a string.
Show me the raw MQTT message wich is sent to awtrix.

daath, 2024-12-30:
Well, `"{% set snut = states('sensor.snustats_today') ~ '​' %}{{ snut }}"` works but `"{% set snut = states('sensor.snustats_today') | string %}{{ snut }}"` does not ...

Blueforcer, 2024-12-30:
> Well, `"{% set snut = states('sensor.snustats_today') ~ '​' %}{{ snut }}"` works but `"{% set snut = states('sensor.snustats_today') | string %}{{ snut }}"` does not ...

Connect to your broker with an mqtt client like the MQTT explorer and see if the text is a string or not.
using | string can sometimes introduce subtle formatting or type issues in certain contexts. Essentially, | string may not always behave the same way as direct concatenation, so it can cause unexpected results if the receiving end  expects a plain string without extra formatting.

daath, 2024-12-30:
I'm using the HACS integration, which does the MQTT communication. He may need to fix his message sending. I just had a look at the code, and it looks like he just sends the payload without doing anything, so it should be fixable there.

@Turtlepaw are you using the HACS-integration?

Blueforcer, 2024-12-30:
i can also cast it to a string in the next verison. However, it is not a bug.

daath, 2024-12-30:
> i can also cast it to a string in the next verison. However, it is not a bug.

It might be a good idea, but I think it should be the HACS-integration that fixes this, as it is a side effect of how he sends MQTT-messages - he uses json.dumps on the payload, which detects ints and floats and casts them as python "numbers" ... I proposed a fix to him (I am not a python programmer, so it was basically only an idea) ...

Turtlepaw, 2024-12-30:
> @Turtlepaw are you using the HACS-integration?

@daath, yes I am using the same integration.

daath, 2024-12-30:
> > @Turtlepaw are you using the HACS-integration?
> 
> @daath, yes, I am using 10der/homeassistant-custom_components-awtrix

OK, that is our problem. I created a pull request to fix this - while you wait for a merge, in ´custom_components/awtrix/awtrix.py´ you can change:

```
            if 'icon' in msg:
                if str(msg["icon"]).startswith(('http://', 'https://')):
                    icon = await self.hass.async_add_executor_job(getIcon, str(msg["icon"]))
                    if icon:
                        msg["icon"] = icon
                
            payload = json.dumps(msg) if len(msg) else ""
```

to:

```
            if 'icon' in msg:
                if str(msg["icon"]).startswith(('http://', 'https://')):
                    icon = await self.hass.async_add_executor_job(getIcon, str(msg["icon"]))
                    if icon:
                        msg["icon"] = icon

            if 'text' in msg:
                msg['text'] = str(msg['text'])
                
            payload = json.dumps(msg) if len(msg) else ""
```

It's around line 56.

Turtlepaw, 2024-12-30:
> OK, that is our problem. I created a pull request to fix this - while you wait for a merge, in ´custom_components/awtrix/awtrix.py´ you can change:
>
> ```
>             if 'icon' in msg: [...]

I just currently append a zero-width space to the number and that works perfectly:

```yaml
text: >
  {% set steps = states('sensor.google_pixel_watch_daily_steps_4') | int %}
  {{ steps ~ '​' }}  # Appending zero-width space
```
*Minimal working code*

<details>
  <summary>📸 Awtrix Preview</summary>
  <p>

[image]
</p>
</details>

<details>
  <summary>🤖 Home Assistant Automation</summary>
  <p>

This automation uses a third-party library: [10der/homeassistant-custom_components-awtrix](https://github.com/10der/homeassistant-custom_components-awtrix)

```yaml
alias: Awtrix Steps
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.google_pixel_watch_daily_steps_2
  - trigger: time_pattern
    minutes: "14"
conditions: []
actions:
  - action: awtrix.awtrix_2d8db4_push_app_data
    data:
      name: Steps
      data:
        text: >
          {% set steps = states('sensor.google_pixel_watch_daily_steps_2') | int
          %} {{ "{:,.0f}".format(steps).replace(',', '.') ~ '​' }}
        icon: "30415"
        duration: 0
        pushIcon: 0
        lifetime: 900
        repeat: 1
mode: single
```

</p>
</details>

Since we've figured out that it's an issue with the custom component, I'll close this now.

closed 2024-12-30 by Turtlepaw