Blueforcer/awtrix3 #620: Help needed, CustomApps coding, Documentation

itskovacs, 2024-10-21:
Afternoon,

I have been playing with Aw3 for a while now, and I would like to go a bit deeper into Apps after using MQTT and API without issue.

I noticed a folder called _CUSTOMAPPS_ in the files, is there any way to actually code an App and create the json in the folder to use the newly created App. By App, I mean an App with a real program, just like any uploading an Arduino code, not just a "simple" json like displayed in the documentation
```
{
  "text": "Hello, AWTRIX 3!",
  "rainbow": true,
  "duration": 10
}
```

But a complex one, the case is for a Timer app. I would like to code the application to display what I want, and it requires to code.
Is there any way to do so ?

If it's not useable, might be worth the shot to indicate in the docs that this folder is not to be use for custom apps, nothing much on internet except for Awtrix 2.

Thanks,
Have a good evening

Blueforcer, 2024-10-21:
AWTRIX can't handle any app logic by itself. It's not designed for that. It's a Smarthome companion.
The customapp folder is for Jsons which are sent with the "save" Tag. AWTRIX will load it after boot. This is for static apps like welcome messages or full screen gifs wich doesn't need any or many updates. So you doesn't need to send it again every time Awtrix reboots.

itskovacs, 2024-10-21:
Thank you for these clarifications ! I'll use a regular py script to handle this task

closed 2024-10-21 by itskovacs

Blueforcer, 2024-10-21:
I will add this note to the documentation